DanielOgorchock/joycond #2: Add support for androids ueventd lib

bylaws, 2019-06-15:
Unlike most Linux distros android uses its own uevent library,  this should be supported.

Putting here to track my progress.

BTW, when I tried to use the joycon driver with android I never got any subcmd reply reports sent back as reply to calib read, I just get a barrage of 0x3f packets.

DanielOgorchock, 2019-06-15:
That's funny timing. I just replied to your mailing list email. I think the calibration timeout is due to the subcommands being sent in the probe. Deferring all of that to run outside the probe might make it work on Android. Worth a shot at least, since it seems like it isn't ever sending the subcommand.

DanielOgorchock, 2019-10-17:
I've just refactored the daemon in preparation to add more functionality (including support potentially for the rail driver). Part of that refactor involved isolating all the udev code to one file, so we should have an easier time support uevent when it comes to adding android support. Do you know if there's a C or C++ library for uevent integration available?

closed 2019-11-09 by bylaws