DanielOgorchock/joycond #3: [Feature request] Hide single joycons when combined

parkerlreed, 2019-09-20:
The single joycon inputs are killed which is nice but SDL still sees them first meaning games will try and use those "dead" inputs.

I had to manually remove /dev/input/event5 and 6 to have SDL only see the combined. Then it worked fine in games with the combined mapped in gamecontrollerdb.txt

```
INFO: There are 3 joysticks attached
INFO: Joystick 0: Nintendo Switch Left Joy-Con
INFO:        axes: 4
INFO:       balls: 0
INFO:        hats: 0
INFO:     buttons: 18
INFO: instance id: 0
INFO:        guid: 050000007e0500000620000001000000
INFO:     VID/PID: 0x057e/0x2006
INFO: Joystick 1: Nintendo Switch Right Joy-Con
INFO:        axes: 4
INFO:       balls: 0
INFO:        hats: 0
INFO:     buttons: 18
INFO: instance id: 1
INFO:        guid: 050000007e0500000720000001000000
INFO:     VID/PID: 0x057e/0x2007
INFO: Joystick 2: Nintendo Switch Combined Joy-Cons
INFO:        axes: 4
INFO:       balls: 0
INFO:        hats: 0
INFO:     buttons: 18
INFO: instance id: 2
INFO:        guid: 050000007e0500000620000001000000
INFO:     VID/PID: 0x057e/0x2006
INFO: Watching joystick 2: (Nintendo Switch Combined Joy-Cons)
INFO: Joystick has 4 axes, 0 hats, 0 balls, and 18 buttons
INFO: ```

DanielOgorchock, 2019-09-22:
I just pushed a new patch. It adds a udev rule which prevents userspace from accessing the physical controllers. Joycond changes the permissions itself for the virtual combined controllers or the passthrough single controllers. Can you test if this works for you?

parkerlreed, 2019-09-23:
Can check in a second. Thanks.

parkerlreed, 2019-09-23:
I can't seem to get them to combine now. They both connect and when I press the triggers they never show the same lights.

Is the udev rule possibly blocking joycond from seeing them?

DanielOgorchock, 2019-09-23:
Is joycond being run as root? Do the controller player LEDs blink in unison? If they're blinking joycond is seeing them.

parkerlreed, 2019-09-23:
Just noticed there was some dmesg output

```
[  446.746272] joycon 0005:057E:2006.0008: remove
[  446.825345] joycon 0005:057E:2006.0008: setting player leds
[  446.825361] joycon 0005:057E:2006.0008: Failed to send output report ret=-5
[  446.825369] joycon 0005:057E:2006.0008: send subcommand failed; ret=-5
[  446.825376] leds 0005:057E:2006.0008:player4: Setting an LED's brightness failed (-5)
[  446.825438] joycon 0005:057E:2006.0008: setting player leds
[  446.825447] joycon 0005:057E:2006.0008: Failed to send output report ret=-5
[  446.825453] joycon 0005:057E:2006.0008: send subcommand failed; ret=-5
[  446.825459] leds 0005:057E:2006.0008:player3: Setting an LED's brightness failed (-5)
[  446.825625] joycon 0005:057E:2006.0008: setting player leds
[  446.825638] joycon 0005:057E:2006.0008: Failed to send output report ret=-5
[  446.825645] joycon 0005:057E:2006.0008: send subcommand failed; ret=-5
[  446.825652] leds 0005:057E:2006.0008:player2: Setting an LED's brightness failed (-5)
[  446.825792] joycon 0005:057E:2006.0008: setting player leds
[  446.825802] joycon 0005:057E:2006.0008: Failed to send output report ret=-5
[  446.825808] joycon 0005:057E:2006.0008: send subcommand failed; ret=-5
[  446.825814] leds 0005:057E:2006.0008:player1: Setting an LED's brightness failed (-5)
```

parkerlreed, 2019-09-23:
Here's the full log. 
[out.txt](https://github.com/DanielOgorchock/joycond/files/3640598/out.txt)

DanielOgorchock, 2019-09-24:
Those -5 return codes are showing up after the hid device has been removed, so it looks like the controller was disconnected or powered off.

parkerlreed, 2019-09-24:
Hmm weird. It was working on the previous version. Ill do some more testing

DanielOgorchock, 2019-09-24:
It might be helpful to look at the joycond output to see if it detects the controllers or not with `sudo journalctl -u joycond --follow`

parkerlreed, 2019-09-24:
Ok so new version seems to segfault

```
           PID: 6248 (joycond)
           UID: 0 (root)
           GID: 0 (root)
        Signal: 11 (SEGV)
     Timestamp: Mon 2019-09-23 20:53:29 EDT (41s ago)
  Command Line: /usr/bin/joycond
    Executable: /usr/bin/joycond
 Control Group: /system.slice/joycond.service
          Unit: joycond.service
         Slice: system.slice
       Boot ID: c2fc216ee09d4b36bc2275809fc903a4
    Machine ID: 2480fa6dd54744e2b1478b12d1d5bc8a
      Hostname: alarm
       Storage: /var/lib/systemd/coredump/core.joycond.0.c2fc216ee09d4b36bc2275809fc903a4.6248.1569286409000000000000.lz4
       Message: Process 6248 (joycond) of user 0 dumped core.
                
                Stack trace of thread 6248:
                #0  0x0000007f7c671c20 _ZNKSt12__basic_fileIcE7is_openEv (libstdc++.so.6)
                #1  0x0000005567fd0398 n/a (joycond)
                #2  0x0000005567fd04dc n/a (joycond)
                #3  0x0000005567fcd4b4 n/a (joycond)
                #4  0x0000005567fcd9fc n/a (joycond)
                #5  0x0000005567fcbf48 n/a (joycond)
                #6  0x0000007f7c4476e4 __libc_start_main (libc.so.6)
                #7  0x0000005567fcd298 n/a (joycond)
                #8  0x0000005567fcd298 n/a (joycond)
```

I reverted to the previous commit and it is working as expected.

parkerlreed, 2019-09-24:
I'll try to get a debug build going to get a more useful coredump

parkerlreed, 2019-09-24:
Oddly enough the debug build doesn't crash...

Current commit on debug build works and the controllers are hidden successfully

```
[alarm@alarm test]$ file /usr/bin/joycond 
/usr/bin/joycond: ELF 64-bit LSB pie executable, ARM aarch64, version 1 (GNU/Linux), dynamically linked, interpreter /lib/ld-linux-aarch64.so.1, BuildID[sha1]=c1cb1ec96345bfe6c25bfd93f72f1d302139390e, for GNU/Linux 3.7.0, with debug_info, not stripped
```

parkerlreed, 2019-09-24:
Verdict: It is how I'm building it somehow

From git this works

```
cmake .
make -j4
sudo make install
```

My packing is using some custom directories for the package to be built and also moves /lib to /usr/lib since on Arch Linux /lib is just a symlink to /usr/lib

```
build() {
	cd "$srcdir/joycond"
	cmake -DCMAKE_INSTALL_PREFIX=/usr .
	make
}

package() {
	cd "$srcdir/joycond"
	make DESTDIR="$pkgdir/" install
	mv "$pkgdir/lib" "$pkgdir/usr/"
}
```

This is the one that crashes. Does joycond have any hardcoded paths?

parkerlreed, 2019-09-24:
Here's the backtrace for my package with debug enabled

```
#0  std::__basic_file<char>::is_open (this=this@entry=0xffffff49bf43cec0) at basic_file.cc:254
No locals.
#1  0x000000558f224398 in std::basic_filebuf<char, std::char_traits<char> >::is_open (this=0xffffff49bf43ce50) at /usr/include/c++/8.3.0/fstream:1055
No locals.
#2  std::basic_fstream<char, std::char_traits<char> >::is_open (this=0xffffff49bf43ce38) at /usr/include/c++/8.3.0/fstream:1056
No locals.
#3  phys_ctlr::set_player_led (this=this@entry=0x55bf43cdf0, index=index@entry=-2147483648, on=on@entry=true) at /usr/src/debug/joycond/src/phys_ctlr.cpp:212
No locals.
#4  0x000000558f2244dc in phys_ctlr::set_player_leds_to_player (this=0x55bf43cdf0, player=<optimized out>) at /usr/src/debug/joycond/src/phys_ctlr.cpp:238
        i = -2147483648
#5  0x000000558f2214b4 in add_combined_ctlr (physl=0x55bf43cdf0, physr=0x55bf4524d0, epoll_fd=3) at /usr/include/c++/8.3.0/bits/stl_vector.h:805
        combined = <optimized out>
        ctlr_event = {events = 110081, data = {ptr = 0x7f7b270040 <std::cout>, fd = 2066153536, u32 = 2066153536, u64 = 547527000128}}
        found_slot = false
#6  0x000000558f2219fc in ctlr_event_handler (event_fd=13, epoll_fd=3) at /usr/src/debug/joycond/src/main.cpp:230
        ctlr = 0x55bf43cdf0
        kv = <optimized out>
        __for_range = std::map with 2 elements = {["/devices/70006300.serial/serial0/serial0-0/bluetooth/hci0/hci0:11/0005:057E:2006.0001/input/input5/event5"] = 0x55bf43cdf0, 
          ["/devices/70006300.serial/serial0/serial0-0/bluetooth/hci0/hci0:12/0005:057E:2007.0002/input/input6/event6"] = 0x55bf4524d0}
        __for_begin = <optimized out>
        __for_end = <optimized out>
#7  0x000000558f21ff48 in main (argc=<optimized out>, argv=<optimized out>) at /usr/src/debug/joycond/src/main.cpp:314
        i = <optimized out>
        nfds = <optimized out>
        udev = <optimized out>
        mon = 0x55bf43afe0
        udev_mon_fd = 4
        epoll_fd = 3
        udev_mon_event = {events = 1, data = {ptr = 0x5500000004, fd = 4, u32 = 4, u64 = 365072220164}}
        events = {{events = 1, data = {ptr = 0x550000000d, fd = 13, u32 = 13, u64 = 365072220173}}, {events = 2062943100, data = {ptr = 0x7f7b095000 <databases+112>, fd = 2064207872, u32 = 2064207872, u64 = 547525054464}}, {
            events = 2065045696, data = {ptr = 0x7fec6df230, fd = -328338896, u32 = 3966628400, u64 = 549427474992}}, {events = 2401403752, data = {ptr = 0x5, fd = 5, u32 = 5, u64 = 5}}, {events = 5, data = {ptr = 0x7fec6df230, 
              fd = -328338896, u32 = 3966628400, u64 = 549427474992}}, {events = 2401480856, data = {ptr = 0x558f23b008, fd = -1893486584, u32 = 2401480712, u64 = 367473700872}}, {events = 2401403716, data = {ptr = 0x7fec6df270, 
              fd = -328338832, u32 = 3966628464, u64 = 549427475056}}, {events = 2062857868, data = {ptr = 0x558f228300 <__libc_csu_init>, fd = -1893563648, u32 = 2401403648, u64 = 367473623808}}, {events = 0, data = {ptr = 0x7fec6df270, 
              fd = -328338832, u32 = 3966628464, u64 = 549427475056}}, {events = 2062857896, data = {ptr = 0x558f228300 <__libc_csu_init>, fd = -1893563648, u32 = 2401403648, u64 = 367473623808}}}
```

DanielOgorchock, 2019-09-24:
Strange, looks like the player LED index is a garbage number when it's trying to set the LEDs. I doubt this is due to your packaging switching the udev rules install path. Is it maybe changing the compiler optimization levels compared to the default?

DanielOgorchock, 2019-09-24:
Okay, I just pushed a patch that might fix it. I was missing a return value for that set_leds function which is supposed to return a bool.

parkerlreed, 2019-09-24:
Success! Thanks for all the help and confirming I wasn't going crazy (well I am in other ways but eh :D)

closed 2019-09-24 by parkerlreed