DanielOgorchock/joycond #12: How to run this in manjaro

sorryusernameisalreadytaken, 2020-01-08:
Hey there. I try to connect one or 2 as combined joycon, without success. :/
AUR: https://aur.archlinux.org/packages/joycond-git/
My Manjaro: 

What I do:

1. install your git over the aur with trizen and the default install settings
2. Then I run 
> sudo systemctl enable --now joycond
3. Checked it with
> sudo systemctl status joycond

got:
```● joycond.service - joycond
   Loaded: loaded (/etc/systemd/system/joycond.service; enabled; vendor
preset: disabled)
   Active: active (running) since Wed 2020-01-08 08:47:26 CET; 6min ago
 Main PID: 578 (joycond)
    Tasks: 1 (limit: 4915)
   Memory: 1004.0K
   CGroup: /system.slice/joycond.service
           └─578 /usr/bin/joycond

Jan 08 08:47:26 mymy-manjaro systemd[1]: Started joycond.
Jan 08 08:47:26 mymy-manjaro joycond[578]: adding epoll_subscriber: fd=4 ```

4. When I press ZR + Z nothing happens. The joycon is just blinking. When I hold the button the flashing light just stops after about 6 seconds.

Can you anybody help me, did I do something wrong?

DanielOgorchock, 2020-01-08:
It sounds like you don't have the hid-nintendo kernel driver. It's still in review, but if you want to compile a custom kernel with it I have the patches in my Linux fork.

sorryusernameisalreadytaken, 2020-01-08:
> 
> 
> It sounds like you don't have the hid-nintendo kernel driver. It's still in review, but if you want to compile a custom kernel with it I have the patches in my Linux fork.

THX. Sorry I did not know that.
So I think I just need to follow this: https://github.com/DanielOgorchock/linux/blob/ogorchock/Documentation/admin-guide/README.rst

Press thumbs for me :)

closed 2020-01-08 by sorryusernameisalreadytaken