DanielOgorchock/joycond #13: Sticks not centering very well

parkerlreed, 2020-01-16:
I have checked my controllers Switch side and they appear to be fine. Both sticks with joycond maybe seems too sensitive towards the center?

Here are some values between moving them a few times but letting them auto center.

```
Axes:  0:   995  1:   -96  2: -1238  3:     0 
Axes:  0:  1020  1:  -213  2: -2778  3:     0
Axes:  0:    78  1: -1037  2: -1639  3:     0 
Axes:  0:   442  1: -1174  2: -1163  3:     0
Axes:  0:    78  1: -1203  2: -3393  3:     0
```

These are reported in jstest. Depending on if others are seeing the same issue, I wonder if a driver side deadzone would be needed. Testing on the Nintendo side it seems there may already be one in place.I can see changes when doing small wiggles using joycond but not in the stock Nintendo setup.

DanielOgorchock, 2020-01-16:
I might need to bump the default deadzone size in the driver. You can change it from userspace as well as documented here: https://wiki.archlinux.org/index.php/Gamepad#evdev_API_deadzones

parkerlreed, 2020-01-16:
Ahh thanks. Good to know.

parkerlreed, 2020-01-16:
Doesn't appear to be supported? (Also tried with elevated rights)

```
[parker@alarm ~]$ evdev-joystick --showcal /dev/input/js4
evdev ioctl: Invalid argument
Supported Absolute axes:
```

parkerlreed, 2020-01-16:
I am using the serial rail driver on an actual Switch. I think I used the wrong inedex above (need event) but I can't tell which it is

```
lrwxrwxrwx 1 root root   9 Dec 17 07:44 platform-70006040.serial-event -> ../event8
lrwxrwxrwx 1 root root   9 Dec 17 07:44 platform-70006040.serial-event-joystick -> ../event7
lrwxrwxrwx 1 root root   6 Dec 17 07:44 platform-70006040.serial-joystick -> ../js2
lrwxrwxrwx 1 root root   9 Dec 17 07:44 platform-70006200.serial-event -> ../event6
lrwxrwxrwx 1 root root   9 Dec 17 07:44 platform-70006200.serial-event-joystick -> ../event5
lrwxrwxrwx 1 root root   6 Dec 17 07:44 platform-70006200.serial-joystick -> ../js0
```

DanielOgorchock, 2020-01-16:
Easiest way to tell is probably runing evtest without any arguments. It will list all the evdevs along with their names. If joycond is installed you probably have to run as root to see the actual evdevs from the rail driver (its udev rules get rid of the uaccess tag for them). I'm not positive, but it might also be possible to modify the deadzone on the virtual combined controller's evdev instead of each of the base ones individually.

parkerlreed, 2020-01-16:
Setting it for the combined reports that it worked but Im still seeing the off zero values for combined

```
[parker@alarm ~]$ evdev-joystick --evdev /dev/input/event9 --axis 0 --deadzone 4000
Event device file: /dev/input/event9
Axis index to deal with: 0
New dead zone value: 4000
Trying to set axis 0 deadzone to: 4000
  Absolute axis 0x00 (0) (X Axis) Setting deadzone value to : 4000
    (value: -494, min: -32767, max: 32767, flatness: 4000 (=12.21%), fuzz: 250)
[parker@alarm ~]$ evdev-joystick --evdev /dev/input/event9 --axis 1 --deadzone 4000
Event device file: /dev/input/event9
Axis index to deal with: 1
New dead zone value: 4000
Trying to set axis 1 deadzone to: 4000
  Absolute axis 0x01 (1) (Y Axis) Setting deadzone value to : 4000
    (value: -1518, min: -32767, max: 32767, flatness: 4000 (=12.21%), fuzz: 250)
[parker@alarm ~]$ evdev-joystick --evdev /dev/input/event9 --axis 2 --deadzone 4000
Event device file: /dev/input/event9
Axis index to deal with: 2
New dead zone value: 4000
Trying to set axis 2 deadzone to: 4000
[parker@alarm ~]$ evdev-joystick --evdev /dev/input/event9 --axis 3 --deadzone 4000
Event device file: /dev/input/event9
Axis index to deal with: 3
New dead zone value: 4000
```

```
[parker@alarm ~]$ jstest --normal /dev/input/js4
Driver version is 2.1.0.
Joystick (Nintendo Switch Combined Joy-Cons) has 4 axes (X, Y, Rx, Ry)
and 22 buttons (BtnA, BtnB, BtnX, BtnY, BtnZ, BtnTL, BtnTR, BtnTL2, BtnTR2, BtnSelect, BtnStart, BtnMode, BtnThumbL, BtnThumbR, (null), (null), (null), (null), Btn0, Btn1, Btn2, Btn3).
Testing ... (interrupt to exit)
Axes:  0:   523  1:     0  2: -3192  3:   215 Buttons:  0:off  1:off  2:off  3:off  4:off  5:off  6:off  7:off  8:off  9:off 10:off 11:off 12:off 13:off 14:off 15:off 16:off 17:off 18:off 19:off 20:off 21:off 
```

parkerlreed, 2020-01-16:
Ok I realize the numbering was off vs jstest and now all are at 4000

```
[parker@alarm ~]$ sudo evdev-joystick --showcal /dev/input/event9
Supported Absolute axes:
  Absolute axis 0x00 (0) (X Axis) (value: 1008, min: -32767, max: 32767, flatness: 4000 (=12.21%), fuzz: 250)
  Absolute axis 0x01 (1) (Y Axis) (value: -133, min: -32767, max: 32767, flatness: 4000 (=12.21%), fuzz: 250)
  Absolute axis 0x03 (3) (X Rate Axis) (value: -3594, min: -32767, max: 32767, flatness: 4000 (=12.21%), fuzz: 250)
  Absolute axis 0x04 (4) (Y Rate Axis) (value: 709, min: -32767, max: 32767, flatness: 4000 (=12.21%), fuzz: 250)
```

Still not getting zeros when "centered"

DanielOgorchock, 2020-01-18:
I'm not positive, but it might be up to applications to honor the flatness value rather than the kernel reporting values of 0 via evdev.

parkerlreed, 2020-01-18:
Ahhh thanks. I know I was definitely drifting in gzdoom. It has some of its own deadzone sliders but I'll give it a shot close to zero. If the non zero (or close to) values aren't widespread, feel free to close this if needed.

parkerlreed, 2020-01-18:
Yes! gzdoom deadzone at zero and that works great

closed 2020-03-19 by DanielOgorchock