DanielOgorchock/joycond #14: Not working in arch linux (with custom kernel)

loper12, 2020-02-03:
I cannot connect with bluetooth, as neither the default bluetooth manager in cinnamon(blueberry), blueman, or even bluetoothctl are able to connect to the pro controller as it disconnects. The same happens with joycons. When I connect the pro controller with USB the lights blink but when I try to pair, the lights flash like they normally would when trying to connect with a switch.

I have built a custom linux kernel([using these instructions](https://wiki.archlinux.org/index.php/Kernel/Arch_Build_System)) with [these patches](https://patchwork.kernel.org/project/linux-input/list/?series=222179). I have made sure to enable the HID-nintendo functionality in the config file. HID-nintendo is in my kernel. I can verify that /usr/lib/modules/5.5.1-arch1-1/kernel/drivers/hid/hid-nintendo.ko.xz exists.

I am going to try building hid-nintendo as a module.

loper12, 2020-02-03:
It seems i was pairing wrong when trying to pair via wire. You seem to need to hold it down. However, bluetooth functionality is still not working in any way

DanielOgorchock, 2020-02-04:
Can you check the dmesg output when you're trying to pair via BT?

loper12, 2020-02-06:
<font color="#C4A000">Bluetooth</font><font color="#CC0000">: hci0: corrupted ACL packet</font>
<font color="#C4A000">Bluetooth</font><font color="#CC0000">: hci0: corrupted ACL packet</font>
<font color="#C4A000">Bluetooth</font><font color="#CC0000">: Frame is too short (len 1)</font>
<font color="#C4A000">Bluetooth</font><font color="#CC0000">: hci0: corrupted ACL packet</font>
<font color="#C4A000">Bluetooth</font><font color="#CC0000">: hci0: corrupted ACL packet</font>
<font color="#C4A000">Bluetooth</font><font color="#CC0000">: Frame is too short (len 1)</font>

This is what it prints out

loper12, 2020-02-23:
This seems to have been a problem with bluetooth. I have fixed it by disabling bluetooth autosuspend.

closed 2020-02-23 by loper12

sorryusernameisalreadytaken, 2020-02-26:
> This seems to have been a problem with bluetooth. I have fixed it by disabling autosuspend in the bluetooth kernel module.

Is there a chance to test your kernel with a Manjaro os. I really want to use this joycond but I am not ready to build my own kernel :/
Would be nice.

CCF100, 2020-03-02:
@sorryusernameisalreadytaken If you follow the guide [here](https://wiki.archlinux.org/index.php/Kernel/Traditional_compilation) it's surprisingly easy... Then again this is coming from an Arch user so...

CCF100, 2020-03-18:
Also, there's a Linux Kernel PKGBUILD here that makes it really easy to compile your own kernel with patches, but while the kernel module works like this, I can't get it to interface with joycond...

loper12, 2020-03-26:
@CCF100 is bluetooth unable to connect? then you encountered the same problem as pretty much everyone else. just search up bluetooth autosuspend, and you'll find millions of solution to that problem. if your bluetooth is working, (for other devices) then it might be a different problem that might relate to joycond or hid-nintendo. (also the paches(or you could click on series in a patch to get a single patch containing all of the patches) are [here](https://patchwork.kernel.org/project/linux-input/list/?series=257335) if you want to be able to use the guide. just add it to the resources, and you'll be able to use the patch)