DanielOgorchock/joycond #16: Doesn't seem to interface with kernel module when compiled as a patch

CCF100, 2020-03-18:
hid-nintendo doesn't appear to interface with joycond when compiled as a kernel patch:
Dmesg:
`[ 9037.447580] nintendo 0005:057E:2009.0004: unknown main item tag 0x0
[ 9037.447630] nintendo 0005:057E:2009.0004: hidraw2: BLUETOOTH HID v0.01 Gamepad [Pro Controller] on 08:71:90:31:e4:ca`
joycond jornalctl:
`-- Logs begin at Fri 2020-03-13 14:47:54 CDT, end at Wed 2020-03-18 14:11:11 CDT. --
Mar 18 11:39:41 caleb-gaming-laptop-archlinux systemd[1]: Started joycond.
Mar 18 11:39:41 caleb-gaming-laptop-archlinux joycond[893]: adding epoll_subscriber: fd=4
`

DanielOgorchock, 2020-03-18:
From the dmesg, it doesn't look like hid-nintendo is loading. Did you enable the hid-nintendo Kconfigs when you built the kernel?

DanielOgorchock, 2020-03-18:
You should see output similar to this if it's using the driver:
```
[51895.619225] nintendo 0003:057E:2009.0008: hidraw6: USB HID v81.11 Joystick [Nintendo Co., Ltd. Pro Controller] on usb-0000:0f:00.3-4.2/input0
[51896.666268] nintendo 0003:057E:2009.0008: controller MAC = 98:B6:E9:B1:88:81
[51896.746339] input: Nintendo Switch Pro Controller as /devices/pci0000:00/0000:00:08.1/0000:0f:00.3/usb5/5-4/5-4.2/5-4.2:1.0/0003:057E:2009.0008/input/input30
```

CCF100, 2020-03-21:
This?
`Nintendo Joy-Con and Pro Controller support (HID_NINTENDO) [N/m/?] (NEW) m
`

CCF100, 2020-03-22:
Update, I think I figured it out... Looks like I didn't compile the entire patch, I only downloaded a part of the patch... I'll close this when I've confirmed that was the issue...

CCF100, 2020-03-22:
Yep, that was the issue, I make silly mistakes sometimes, lol

closed 2020-03-22 by CCF100