DanielOgorchock/joycond #19: Can't get it to work.

Bonicgamer, 2020-04-11:
When I connect my Joycons to my computer and press the L & R buttons nothing happens.
I looked and saw it might be trying to look in an invalid directory. When I ran _journalctl_ it   showed: 
> no match found for /sys//devices/_(eccetera)_

So if I'm doing something wrong please tell me.

parkerlreed, 2020-04-12:
Do you have the kernel module?

Bonicgamer, 2020-04-12:
> Do you have the kernel module?

Yes, I got it from the AUR, so I might try to install it manually sometime later today to see if that somehow fixes it.

Bonicgamer, 2020-04-13:
>so I might try to install it manually sometime later today to see if that somehow fixes it.

Well it somehow fixed it.

closed 2020-04-13 by Bonicgamer

Magicrafter13, 2021-03-22:
> Do you have the kernel module?

There's a kernel module??