DanielOgorchock/joycond #20: Inverted axis in horizontal mode

aronszabo, 2020-04-12:
I installed hid-nintendo with dkms and joycond on my manjaro system. Using the joy-cons in horizontal mode (for party games). One of the axes get inverted on each joy-con: Y on the left, X on the right joy-con. In Steam Big Picture mode I got it right using SDL_GamepadBind configuration:

    "SDL_GamepadBind"		
    "050000007e0500000720000001800000,dev:xb1:Nintendo Switch Right Joy-Con,platform:Linux,a:b2,b:b1,y:b3,x:b0,start:b9,guide:b8,leftstick:b10,leftx:a1~,lefty:a0,lefttrigger:b4,righttrigger:b6,
    050000007e0500000620000001800000,dev:xb1:Nintendo Switch Left Joy-Con,platform:Linux,a:b8,b:b9,y:b10,x:b7,start:b0,guide:b5,leftstick:b6,leftx:a1,lefty:a0~,lefttrigger:b2,righttrigger:b4,"

In games the axes remain inverted, despite the other bindings working.

Is it possible to implement an option, where the evdev device gets the already inverted joystick controls (and this way no further configuration would be needed)?

DanielOgorchock, 2020-04-17:
I plan to properly address this issue at some point by having special handling for horizontal mode (right now pairing in horizontal mode behaves identically to single, vertical joycon mode). Joycond will map the axes/buttons properly for the rotated controller.

Aggyle, 2022-12-27:
hey , Daniel, are you still working on this problem or is this project dead?