DanielOgorchock/joycond #24: No wifi after combine the two joycons

ricpelo, 2020-06-08:
I have the following motherboard:

[MSI B450 Gaming Pro Carbon AC](https://es.msi.com/Motherboard/B450-GAMING-PRO-CARBON-AC.html)

I'm using both the wifi & bluetooth adapters from the motherboard.

When I paired the two joycons (left & right) and press the triggers to combine them into one controller, the wifi connection begins to works bad, it renders inusable, however the connection seems to be right according to the NetworkManager tray icon.

When I disconnect any of the joycons, the wifi connection begins to works OK immediately.

DanielOgorchock, 2020-06-08:
Does the issue only occur after pressing the triggers to combine them in joycond, or does it still happen when the two controllers are just connected and blinking. I wouldn't expect joycond to have any impact, since the same amount of bluetooth traffic should be present with or without this daemon.

I can somewhat understand why having multiple controllers paired could degrade wifi performance. The motherboard may be time multiplexing the bluetooth and 2.4GHz WiFi connections on the same antenna. Are you connecting to a 2.4GHz access point. If you have a 5GHz access point available, I'd try that, since that should be using a different antenna.

ricpelo, 2020-06-08:
> Does the issue only occur after pressing the triggers to combine them in joycond, or does it still happen when the two controllers are just connected and blinking?

It still happen when the two controllers (not only one) are just connected and blinking. And I just realized that the issue is still reproducible even without the daemon running.

> I can somewhat understand why having multiple controllers paired could degrade wifi performance. The motherboard may be time multiplexing the bluetooth and 2.4GHz WiFi connections on the same [...]

You're right: if I connect to a 5 GHz access point, I can't reproduce this issue. Unfortunately, my main access point is 2.4 GHz and I can't change it because of the Internet service provider.

DanielOgorchock, 2020-06-10:
Closing since this issue seems to be a problem with the MB's bluetooth/wifi time sharing rather than joycond itself.

closed 2020-06-10 by DanielOgorchock