DanielOgorchock/joycond #25: How to uninstall?

PowerUser64, 2020-06-27:
I have done a bit of research, but is unclear how to uninstall this. When I installed it, I followed the instructions in the readme.

greasysock, 2020-06-28:
What I am assuming is, you need to reverse the install steps found in this file: [CMakeLists.txt](../blob/master/CMakeLists.txt), but you need to stop and disable the daemon first. There currently is no uninstall target.

PowerUser64, 2020-06-28:
Alright, sounds good. Thanks!

closed 2020-06-28 by PowerUser64

PowerUser64, 2020-07-01:
Just figured out you can install and uninstall from the AUR and that will remove it

benhriv, 2021-10-14:
> What I am assuming is you need to reverse the install steps found in this file: [CMakeLists.txt](../blob/master/CMakeLists.txt), but you need to stop and disable the daemon first. There currently is no uninstall target.

Sorry, I don't understand. How do you do this?

subtlecobra3864, 2022-01-02:
> What I am assuming is you need to reverse the install steps found in this file: [CMakeLists.txt](../blob/master/CMakeLists.txt), but you need to stop and disable the daemon first. There currently is no uninstall target.

Can you explain this a bit further?

PowerUser64, 2022-01-03:
If you look at the bottom of the file where the lines that read `install()` are, you can delete manually delete the files listed.
[Example:](https://github.com/DanielOgorchock/joycond/blob/master/CMakeLists.txt#L28)
```cmake
install(FILES systemd/joycond.service DESTINATION /etc/systemd/system)
```
means you remove the file `joycond.service` at `/etc/systemd/system`, like this:
```bash
rm /etc/systemd/system/joycond.service
```
and then you can repeat that for all the other

cr1za, 2025-03-10:
> Just figured out you can install and uninstall from the AUR and that will remove it

What is AUR?