DanielOgorchock/joycond #26: [Feature Request] Auto pair Pro controllers

shadowofdarkness, 2020-07-19:
Considering that Pro controllers don't have multiple configurations like joycons would it be possible to remove the L+R requirement for them?

DanielOgorchock, 2020-07-24:
Maybe there should be a config file the daemon reads to determine this behavior. I've been aiming for the default behavior to be similar to pairing the controllers on the switch itself, so that the player order can be set regardless of device probe order.

shadowofdarkness, 2020-07-24:
A config file would be a good way to deal with it. I'm sure a lot of people would like the option since not everyone is going to play local multiplayer games on a computer and would enjoy the ability to skip a step.

nicman23, 2020-08-02:
another way is to only have the L+R combo required when there are multiple controllers controllers connected.

nicman23, 2020-08-14:
i made a branch that auto-pairs pro controllers at https://github.com/nicman23/joycond/tree/autoPair

DanielOgorchock, 2020-08-14:
> i made a branch that auto-pairs pro controllers at https://github.com/nicman23/joycond/tree/autoPair

Cool. I'll plan on adding some config file support tonight or tomorrow to allow us to gate this feature and the hidraw permissions one with it. I think we're going to end up wanting more optional features going forward.

nicman23, 2020-08-14:
just FYI it is pretty rudimentary. if it sees a pro controller that has the status of pairing it just goes ahead and adds it.