DanielOgorchock/joycond #28: How to work on system doesn't use systemd?

leslielg, 2020-08-25:
I'm using MX Linux KDE version, by default it boot without systemd, and this way is recommended by developers from MX team. Could you make joycond working on such system?

DanielOgorchock, 2020-08-26:
All that should be needed is an init script for whatever init system MX Linux KDE uses. AboveGojira had posted an OpenRC init script in this issue: https://github.com/DanielOgorchock/joycond/issues/21

I guess we should start merging the scripts for the various init systems into the repo and have a cmake flag to select (or maybe automatically detect) which init system is in use at time of install.

leslielg, 2020-08-26:
IIRC the hid nintendo driver install also requires systemd, if it detects systemd absence, it will report an error.

DanielOgorchock, 2020-08-26:
hid-nintendo doesn't care about which init system is in use. Do you have an example of the error output you can show me?

leslielg, 2020-08-26:
Here's the output when I try install hid-nintendo, that Chinese string means host closed:
leslie@leslie-desktop ~/Games/dkms-hid-nintendo (master)$ sudo dkms install nintendo -v 2.0

hid-nintendo.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.6.0-2-amd64/updates/

depmod...
System has not been booted with systemd as init system (PID 1). Can't operate.
Failed to connect to bus: 主机关闭

DKMS: install completed.
leslie@leslie-desktop ~/Games/dkms-hid-nintendo (master)$

leslielg, 2020-08-26:
This is the only way I know to install hid-nintendo driver and it is introduced from
https://joaorb64.github.io/2020/04/08/using-nintendo-switch-controllers-on-linux.html

parkerlreed, 2020-08-26:
Neither the driver nor the dkms installer seem to require systemd. There may be something misconfigured with your system in respect to dkms.

parkerlreed, 2020-08-26:
It does report install succeeded. What happens if you try loading the module? The systemd error seems benign

leslielg, 2020-08-28:
not tried yet, how to mannually set joycond start on boot in MX KDE system?

Artturin, 2020-09-03:
you'd have to write a sysvinit service

zen253230, 2022-05-20:
Here is a script I adapted from the internet which runs well on my devuan
You have to make it executable and copy to /etc/init.d

``
#! /bin/sh

# Using the lsb functions to perform the operations.
. /lib/lsb/init-functions
# Process name ( For display )
NAME=joycond
# Daemon
DAEMON=/usr/sbin/joycond
# pid file for the daemon
PIDFILE=/var/run/joycond.pid

# If the daemon is not there, then exit.
test -r "$DAEMON" || exit 5

case $1 in
 start)
  # Checked the PID file exists and check the actual status of process
  if [ -e $PIDFILE ]; then
   status_of_proc -p $PIDFILE $DAEMON "$NAME process" && status="0" || status="$?"
   # If the status is SUCCESS then don't need to start again.
   if [ $status = "0" ]; then
    exit # Exit
   fi
  fi
  # Start the daemon.
  log_daemon_msg "Starting the process" "$NAME"
  # Start the daemon with the help of start-stop-daemon
  # Log the message appropriately
  if start-stop-daemon --start --quiet --oknodo --make-pidfile --pidfile $PIDFILE --exec $DAEMON --background; then
   log_end_msg 0
  else
   log_end_msg 1
  fi
  ;;
 stop)
  # Stop the daemon.
  if [ -e $PIDFILE ]; then
   status_of_proc -p $PIDFILE $DAEMON "Stoppping the $NAME process" && status="0" || status="$?"
   if [ "$status" = 0 ]; then
    start-stop-daemon --stop --quiet --oknodo --pidfile $PIDFILE
    /bin/rm -rf $PIDFILE
   fi
  else
   log_daemon_msg "$NAME process is not running"
   log_end_msg 0
  fi
  ;;
 restart)
  # Restart the daemon.
  $0 stop && sleep 2 && $0 start
  ;;
 status)
  # Check the status of the process.
  if [ -e $PIDFILE ]; then
   status_of_proc -p $PIDFILE $DAEMON "$NAME process" && exit 0 || exit $?
  else
   log_daemon_msg "$NAME Process is not running"
   log_end_msg 0
  fi
  ;;
 reload)
  # Reload the process. Basically sending some signal to a daemon to reload
  # it configurations.
  if [ -e $PIDFILE ]; then
   start-stop-daemon --stop --signal USR1 --quiet --pidfile $PIDFILE --name $NAME
   log_success_msg "$NAME process reloaded successfully"
  else
   log_failure_msg "$PIDFILE does not exists"
  fi
  ;;
 *)
  # For invalid arguments, print the usage message.
  echo "Usage: $0 {start|stop|restart|reload|status}"
  exit 2
  ;;
esac