DanielOgorchock/joycond #30: Individual joycons not merged

ademlabs, 2020-09-13:
Joycond doesn't provide opportunity to merge two individual joycons. 
hid_nintendo is loaded, player light 1 & 2 showing on the respective joycons. Load joycond via `sudo joycond` and the lights blink and go back to 1 & 2. There is no way to choose the pairing mode, so unable to combine both joycons.

The messages shown are:
```
adding epoll_subscriber: fd=4
Creating new phys_ctlr for /dev/input/event3
chmod failedFound Left Joy-Con
driver_name: Nintendo Switch Left Joy-Con
MAC: 98:B6:E9:83:F0:45
adding epoll_subscriber: fd=6
Lone controller paired
Creating new phys_ctlr for /dev/input/event31
chmod failedFound Right Joy-Con
driver_name: Nintendo Switch Right Joy-Con
MAC: 98:B6:E9:58:94:5A
adding epoll_subscriber: fd=16
Lone controller paired
```

Looks like there is a `chmod failed` between creating a new controller and the device detection but I don't have any idea what could be wrong besides that.

DanielOgorchock, 2020-09-13:
Are you pressing any of the joy-cons' triggers during this process?

ademlabs, 2020-09-13:
I've tried all sorts of combination but nothing works. I tried by having joycond loaded as a systemd service as well but at no point do I see the blinking lights to allow me to choose the mode by using the triggers.

ademlabs, 2020-09-13:
Well, it seems that the AUR package is buggy somehow. I rebuilt the binary from the latest source now and it works perfectly. I don't see the `chmod failed` error anymore:

```
adding epoll_subscriber: fd=4
Creating new phys_ctlr for /dev/input/event3
Found Left Joy-Con
driver_name: Nintendo Switch Left Joy-Con
MAC: 98:B6:E9:83:F0:45
adding epoll_subscriber: fd=5
Creating new phys_ctlr for /dev/input/event31
Found Right Joy-Con
driver_name: Nintendo Switch Right Joy-Con
MAC: 98:B6:E9:58:94:5A
adding epoll_subscriber: fd=14
Waiting controller needs partner
Found right
Waiting controller needs partner
Found left
adding epoll_subscriber: fd=25
Creating combined joy-con input
```

closed 2020-09-13 by ademlabs

gajama, 2022-01-12:
i know this issue is closed, but I found it while looking for what was causing chmod Failed messages in systemd, as I couldn't get virtual controller pairing to work.
I was also using the AUR package, and so I tried building from source, but that made no difference.
What did work, was removing ```User=root``` from the Systemd Unit file.