DanielOgorchock/joycond #36: " cmake . " command not working

UsernameNotAvalible, 2020-10-11:
when i type `cmake .` i get an error that says

`CMake Error at CMakeLists.txt:1 (cmake_minimum_required):
  CMake 3.13 or higher is required.  You are running version 3.10.2

-- Configuring incomplete, errors occurred!
`

theofficialgman, 2020-10-14:
the error tells you the issue: CMake 3.13 or higher is required.  You are running version 3.10.2
you need to update to a newer version of cmake (either through a package or building it yourself)

zzorotus, 2020-10-29:
I got a similar error:
```
-- Checking for module 'libevdev'
--   No package 'libevdev' found
CMake Error at /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:463 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:643 (_pkg_check_modules_internal)
  CMakeLists.txt:10 (pkg_check_modules)

-- Configuring incomplete, errors occurred!

```
I'm currently running CMake 3.16

FranklinsHand, 2020-12-01:
> I got a similar error:
>
> ```
> -- Checking for module 'libevdev'
> --   No package 'libevdev' found
> CMake Error at /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:463 (message):
>   A required package [...]

I had to run `$ sudo apt install libevdev-dev` and `$ sudo apt install libudev-dev` in order to get it to work. There are others who also had to run `$ sudo apt-get install pkg-config`, just in case you run into an error with that as well.

codefist, 2021-03-07:
same problem here.  Pop OS.  cmake just cannot find libevdev.

NK308, 2021-06-27:
This actually should just be added to the README and we're fine.