DanielOgorchock/joycond #39: IMU devices from latest kernels are not hidden (Bluetooth)

parkerlreed, 2020-10-20:
Similar to the previous issue I reported back when the single joycons were showing up before combined, the IMU are now showing up before the combined when using Bluetiooth.

Maybe hide these as well?

```
INFO: There are 3 joysticks attached
INFO: Joystick 0: Nintendo Switch Left Joy-Con IMU
INFO:        axes: 6
INFO:       balls: 0
INFO:        hats: 0
INFO:     buttons: 0
INFO: instance id: 0
INFO:        guid: 050000007e0500000620000001800000
INFO:     VID/PID: 0x057e/0x2006
INFO: Joystick 1: Nintendo Switch Right Joy-Con IMU
INFO:        axes: 6
INFO:       balls: 0
INFO:        hats: 0
INFO:     buttons: 0
INFO: instance id: 1
INFO:        guid: 050000007e0500000720000001800000
INFO:     VID/PID: 0x057e/0x2007
INFO: Joystick 2: Nintendo Switch Combined Joy-Cons
INFO:        axes: 4
INFO:       balls: 0
INFO:        hats: 0
INFO:     buttons: 22
INFO: instance id: 2
INFO:        guid: 060000007e0500000820000000000000
INFO:     VID/PID: 0x057e/0x2008
INFO:
```

DanielOgorchock, 2020-10-21:
Is this while using the dkms module? If so it doesn't include changes to joydev in the kernel to blacklist the IMU input devs from joydev.

DanielOgorchock, 2020-10-21:
If this is something using the evdevs then I'm not sure of a great solution. Removing userspace access to the IMU devs prevents them from being used by programs like dolphin.

parkerlreed, 2020-10-21:
This is on the Switch running L4T (Arch). The kernel has the nintendo module built-in.

parkerlreed, 2020-10-21:
Even if not upstream, is there a udev file I can create to hide them for my use case?

DanielOgorchock, 2020-10-21:
Uaccess for the IMUs are added by this joycond udev rule. https://github.com/DanielOgorchock/joycond/blob/master/udev/72-joycond.rules

Should be able to comment out those lines or add another rule to -=uaccess

parkerlreed, 2020-10-21:
Commenting those out does not hide them. I'll see what can can figure out with repsect to permissions and also testing the second option of -uaccess after those

parkerlreed, 2020-10-21:
Ok so SDL is using evdev. If I chmod 600 both the event* devices it hides them properly.