DanielOgorchock/joycond #40: "Cmake ." command not working!

UsernameNotAvalible, 2020-11-01:
When i try to run "cmake ." i get an error that says:

```
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
CMake Error at /usr/local/share/cmake-3.19/Modules/FindPkgConfig.cmake:609 (message):
  pkg-config tool not found
Call Stack (most recent call first):
  /usr/local/share/cmake-3.19/Modules/FindPkgConfig.cmake:741 (_pkg_check_modules_internal)
  CMakeLists.txt:10 (pkg_check_modules)

CMake Error at /usr/local/share/cmake-3.19/Modules/FindPkgConfig.cmake:609 (message):
  pkg-config tool not found
Call Stack (most recent call first):
  /usr/local/share/cmake-3.19/Modules/FindPkgConfig.cmake:741 (_pkg_check_modules_internal)
  CMakeLists.txt:11 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "/home/elinero/joycond/CMakeFiles/CMakeOutput.log".
```
Please help! i have the newest version of Cmake. the "CMakeOutput.log" file says:

```
The system is: Linux - 5.4.0-51-generic - x86_64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/cc 
Build flags: 
Id flags:  

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/home/elinero/joycond/CMakeFiles/3.19.0-rc2/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /usr/bin/c++ 
Build flags: 
Id flags:  

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"

The CXX compiler identification is GNU, found in "/home/elinero/joycond/CMakeFiles/3.19.0-rc2/CompilerIdCXX/a.out"

Detecting C compiler ABI info compiled with the following output:
Change Dir: /home/elinero/joycond/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_bf954/fast && /usr/bin/make  -f CMakeFiles/cmTC_bf954.dir/build.make CMakeFiles/cmTC_bf954.dir/build
make[1]: Entering directory '/home/elinero/joycond/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_bf954.dir/CMakeCCompilerABI.c.o
/usr/bin/cc   -v -o CMakeFiles/cmTC_bf954.dir/CMakeCCompilerABI.c.o -c /usr/local/share/cmake-3.19/Modules/CMakeCCompilerABI.c
Using built-in specs.
COLLECT_GCC=/usr/bin/cc
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 7.5.0-3ubuntu1~18.04' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-7 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04) 
COLLECT_GCC_OPTIONS='-v' '-o' 'CMakeFiles/cmTC_bf954.dir/CMakeCCompilerABI.c.o' '-c' '-mtune=generic' '-march=x86-64'
 /usr/lib/gcc/x86_64-linux-gnu/7/cc1 -quiet -v -imultiarch x86_64-linux-gnu /usr/local/share/cmake-3.19/Modules/CMakeCCompilerABI.c -quiet -dumpbase CMakeCCompilerABI.c -mtune=generic -march=x86-64 -auxbase-strip CMakeFiles/cmTC_bf954.dir/CMakeCCompilerABI.c.o -version -fstack-protector-strong -Wformat -Wformat-security -o /tmp/ccoxDqU4.s
GNU C11 (Ubuntu 7.5.0-3ubuntu1~18.04) version 7.5.0 (x86_64-linux-gnu)
	compiled by GNU C version 7.5.0, GMP version 6.1.2, MPFR version 4.0.1, MPC version 1.1.0, isl version isl-0.19-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring nonexistent directory "/usr/local/include/x86_64-linux-gnu"
ignoring nonexistent directory "/usr/lib/gcc/x86_64-linux-gnu/7/../../../../x86_64-linux-gnu/include"
[... 354 lines cut ...]
    arg [-L/usr/lib/x86_64-linux-gnu] ==> dir [/usr/lib/x86_64-linux-gnu]
    arg [-L/usr/lib/../lib] ==> dir [/usr/lib/../lib]
    arg [-L/usr/lib/gcc/x86_64-linux-gnu/7/../../..] ==> dir [/usr/lib/gcc/x86_64-linux-gnu/7/../../..]
    arg [CMakeFiles/cmTC_db11b.dir/CMakeCXXCompilerABI.cpp.o] ==> ignore
    arg [-lstdc++] ==> lib [stdc++]
    arg [-lm] ==> lib [m]
    arg [-lgcc_s] ==> lib [gcc_s]
    arg [-lgcc] ==> lib [gcc]
    arg [-lc] ==> lib [c]
    arg [-lgcc_s] ==> lib [gcc_s]
    arg [-lgcc] ==> lib [gcc]
    arg [/usr/lib/gcc/x86_64-linux-gnu/7/crtendS.o] ==> ignore
    arg [/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/crtn.o] ==> ignore
  collapse library dir [/usr/lib/gcc/x86_64-linux-gnu/7] ==> [/usr/lib/gcc/x86_64-linux-gnu/7]
  collapse library dir [/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu] ==> [/usr/lib/x86_64-linux-gnu]
  collapse library dir [/usr/lib/gcc/x86_64-linux-gnu/7/../../../../lib] ==> [/usr/lib]
  collapse library dir [/lib/x86_64-linux-gnu] ==> [/lib/x86_64-linux-gnu]
  collapse library dir [/lib/../lib] ==> [/lib]
  collapse library dir [/usr/lib/x86_64-linux-gnu] ==> [/usr/lib/x86_64-linux-gnu]
  collapse library dir [/usr/lib/../lib] ==> [/usr/lib]
  collapse library dir [/usr/lib/gcc/x86_64-linux-gnu/7/../../..] ==> [/usr/lib]
  implicit libs: [stdc++;m;gcc_s;gcc;c;gcc_s;gcc]
  implicit dirs: [/usr/lib/gcc/x86_64-linux-gnu/7;/usr/lib/x86_64-linux-gnu;/usr/lib;/lib/x86_64-linux-gnu;/lib]
  implicit fwks: []
```

Intellb, 2020-11-30:
CMake Error at /usr/local/share/cmake-3.19/Modules/FindPkgConfig.cmake:609 (message):
  pkg-config tool not found is telling you dont have that package 
Try this:
**sudo apt-get install pkg-config**

FranklinsHand, 2020-12-01:
I had a similar issue today, I am currently running the most up to date version of POP_OS.
I had to install two packages to get 'cmake .' to run.
libudev-dev
libevdev-dev
It took a little bit to figure it out, and it seems I am not the only one unfamiliar with some of these dependencies, and there is little documentation on libevdev online. Would it be overkill to update the documentation to include what packages are necessary?

tr1p0p, 2021-01-18:
> I had a similar issue today, I am currently running the most up to date version of POP_OS.
> I had to install two packages to get 'cmake .' to run.
> libudev-dev
> libevdev-dev
> It took a little bit to [...]

I actually do that in Pop-OS too, and cmake DON'T WANT to see libevdev , say "not found modules" and close...

And... they're actually no clues for solving that.

jakebrake-fumblefuck, 2021-07-11:
might be a little late but i had this issue.
went through a lot of junk info finally tried this 
$ sudo apt-get install -y git make curl unzip gedit automake autocon... 
$ sudo apt install libudev-dev

hope it helps