DanielOgorchock/joycond #48: Export Combined Joycons' LED status

joaorb64, 2021-01-06:
As discussed in https://github.com/joaorb64/joycond-cemuhook/issues/27, joycond-cemuhook could easly find which controllers are part of a virtual Combined Joycon device by checking LED status. Would that be possible? I tried it myself but couldn't manage to do it.

Alternatively, if the combined joycons' uniq is set as one of the controllers' uniq we could get that controller's LED status and find its pair.

nicokoch, 2021-01-06:
The correct way to provide this information is implementing #6 . I don't know if daniel is working on something though

DanielOgorchock, 2021-01-07:
I've really wanted to have the combined controller's uniq being set as both the individual controllers' uniqs delimited by a comma or something. Unfortunately uinput doesn't support setting the device's uniq value. There's a proposed patch on a kernel mailing list from end of 2019 I think that was never merged sadly. This would be the mos idiomatic solution I think.

I've never messed with evdev's LED api. I wouldn't be surprised if uinput doesn't support it either (though I haven't looked into it at all).

I haven't put any work into adding a dbus interface, though it would be nice I think. The quickest solution may just be writing a temp file per combined joycon containing the uniqs of each of its components.

closed 2021-01-08 by joaorb64