DanielOgorchock/joycond #56: Joy Con Rumble not stopping

kibun1, 2021-02-07:
Hi,

While trying to get my joy cons to work while playing Okami HD on Steam, I've come across the problem that the JoyCon (both of them, in fact) start rumbling with maximum power and don't stop until I shut it down.
For me I can cause that to occur just by playing Okami HD and jumping a single time.

It could be a problem with the game, steam, or proton, which are all three involved in this, so if this is not in the fault of this driver, feel free to close the issue. If that's the case, I'd be happy if someone could explain to me if there's a way to switch off rumble completely, though.

Best regards

viluon, 2021-03-19:
Also happens in Steam controller configuration when you select "identify controller" (which should make the chosen controller vibrate a little bit). As in @kibun1's case, the rumble starts with maximum power and never stops.

Spenk009, 2021-09-28:
I have a pair of knock-off switch controllers (shaped differently and only charge via micro-USB). When connecting these via Bluetooth, they immediately start rumbling and won't stop until disconnected. The issue does not persist on USB, where they work flawlessly.

What logs can I attach?

systemctl status joycond
`Sep 28 21:47:43 W541 joycond[998]: Found left
Sep 28 21:47:43 W541 joycond[998]: DEVNAME=event30 ACTION=remove DEVPATH=/devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11:1.0/bluetooth/hci0/hci0:256/0005:>
Sep 28 21:47:50 W541 joycond[998]: Removing /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11:1.0/bluetooth/hci0/hci0:256/0005:057E:2006.000A/input/input48/>
Sep 28 21:47:50 W541 joycond[998]: DEVNAME=event30 ACTION=add DEVPATH=/devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11:1.0/bluetooth/hci0/hci0:256/0005:057>
Sep 28 21:47:50 W541 joycond[998]: Creating new phys_ctlr for /dev/input/event30
Sep 28 21:47:50 W541 joycond[998]: Found Left Joy-Con
Sep 28 21:47:50 W541 joycond[998]: driver_name: Nintendo Switch Left Joy-Con
Sep 28 21:47:50 W541 joycond[998]: MAC: 9C:54:1E:7D:B0:0A
Sep 28 21:47:50 W541 joycond[998]: adding epoll_subscriber: fd=5
Sep 28 21:47:53 W541 joycond[998]: DEVNAME=event30 ACTION=remove DEVPATH=/devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11:1.0/bluetooth/hci0/hci0:256/0005:`