DanielOgorchock/joycond #64: Joycon button assignment problem

GrandUK1, 2021-05-03:
After managing to pair my joycons and having them as one controller i've found that the dpad, triggers and bumpers aren't assigned properly. in xbox cloud gaming (at least) the dpad doesnt seem to be mapped properly and neither do the triggers or bumpers. This is causing some issues within the games i am trying to play. The A and B buttons appear to be swapped as well but isnt as much of a problem as the others as they are just swapped.

Is there anyway to manually change the mapping myself or have i done something to break the daemon? If you need specific logs i can provide.

Lin-Buo-Ren, 2021-09-28:
I have this problem as well with the swapped A/B keys, it would be preferable if joycond can be configured to handle that.

Lin-Buo-Ren, 2021-09-29:
For the swapped A/B button issue I've created a workaround here: https://github.com/Lin-Buo-Ren/joycond/commit/e5a32b4d866f93e8f83af972729a7619ac39b8c9