DanielOgorchock/joycond #67: Validate that controllers work correctly?

hildebro, 2021-05-14:
I've installed `dkms-hid-nintendo` and `joycond` as described in the respective repos without any error. 
I connected my switch pro controller and it paired into player slot 1 without me having to press L+R. 
But it doesn't look like the controller is actually working. In steam, no controller shows up. And Dolphin just shows `[disconnected] evdev/0/Nintendo Pro Controller` in the emulated wii controller section. My `joycond` journal is pretty empty:
```
-- Journal begins at Sat 2021-02-13 03:18:33 CET. --
May 14 15:36:10 x systemd[1]: Started joycond.
May 14 15:36:10 x joycond[67977]: adding epoll_subscriber: fd=4
May 14 15:44:32 x systemd[1]: Stopping joycond...
May 14 15:44:32 x systemd[1]: joycond.service: Deactivated successfully.
May 14 15:44:32 x systemd[1]: Stopped joycond.
May 14 15:44:32 x systemd[1]: Started joycond.
May 14 15:44:32 x joycond[74185]: adding epoll_subscriber: fd=4
```

I can't pin down, whether this is a joycond issue or a steam/dolphin issue. Is there a way to validate that linux actually recognized the controller?

Ultracoolguy, 2021-06-08:
Use `evtest`.

hvalim, 2021-11-15:
I am having the same problem; the Pro Controller appears to be connected correctly and the installation proceeded without any errors, but Steam does not recognize it at all. Using `evtest` I also can see that it responds to input and the input is recognized.