DanielOgorchock/joycond #68: Cannot get Cmake . to work On Fedora

Dokorege, 2021-07-12:
Note: (Sorry if this issue is listed in a strange way, I'm unfamiliar with github and it's Comment Structure ha ha.)

[image]

**Cmake will output something along the lines of this: 

-- Checking for module 'libevdev'
-- Package 'libevdev', required by 'virtual:world', not found
CMake Error at /usr/share/cmake/Modules/FindPkgConfig.cmake:556 (message):
A required package was not found
Call Stack (most recent call first):
/usr/share/cmake/Modules/FindPkgConfig.cmake:778 (_pkg_check_modules_internal)
CMakeLists.txt:10 (pkg_check_modules)**

Installed the needed package as described in the error, multiple times, restarted my pc, updated repositories, etc 

I honestly have no idea if it's cmake . that is not detecting the package I've downloaded or if I'm doing something wrong (Which is most likely the case because I'm new to fedora.) 

I've been trying to find solutions for this problem for hours now and yet I cannot find anything even remotely related to Fedora when it comes to this software. I've seen different solutions which can all essentially be summed down to "sudo apt-get install libudev-dev" (Which of course is only related to people who uses Ubuntu and other distros of the sort) Which as I stated before I attempted to download the necessary packages and after doing so I still get the same error message. 

For sake of Haste I will be glad to display the Output log file, but within another Comment.

parkerlreed, 2021-07-12:
Did you also install `libevdev-devel`?

Dokorege, 2021-07-12:
> Did you also install `libevdev-devel`?

Yep, Everything I found pointed to downloading the libevdev and libevdev-devel packages, still does not work unfortunately.

Dafnik, 2021-07-28:
Hi, after installing `libevdev-devel` I found the second package which was missing. `systemd-devel` or `libudev-devel`.

I hope this helps.

PR #72 (2021-07-30): Add Fedora install instructions
Fixes #68

For joycond to work on Fedora you need `libevdev-devel` and 'libudev-devel` installed.
commit f9a66914 (2021-07-30): Add Fedora install instructions

Fixes #68
closed 2021-07-30 by DanielOgorchock