DanielOgorchock/joycond #69: Virtual controller support worse than steam hijacking

clarfonthey, 2021-07-25:
Not sure whether to file this as a Steam bug or a bug here, but considering how Valve doesn't seem to respond to Switch controller issues, I figured I'd start here.

I tried updating to the new joycond and it works in one sense, where Valve no longer seriously messes up the status of the Pro controller outside of Steam. Unfortunately, the detection still has a few issues, namely:

1. It's only recognised as a generic gamepad, not a Switch controller. Not sure if this is Valve's fault or if the new name is what's tripping it up.
2. It triggers bugs I used to have in Proton which were since fixed, where anything that triggers vibration will lock the controller in a vibrating state and prevent it from being recognised until I unplug it and plug it back in.
3. The README seems to imply that you can have the virtual and non-virtual pairing simultaneously, but at least from what I can tell, it's only one or the other. Might be worth clarifying that.

Honestly, I'm seriously questioning whether it's really worth it to add these kinds of workarounds since it seems that no matter what, there's going to be something breaking until Valve properly supports the driver, but 🤷🏻.

DanielOgorchock, 2021-07-25:
Does https://github.com/nicman23/dkms-hid-nintendo/pull/48 help with the rumble related issues?

clarfonthey, 2021-07-25:
Based upon the changes, it looks like it actually probably might. Will get back to you tomorrow.

Also changed the title of the issue to be less harsh; I was mostly frustrated when I wrote it, which is probably not the best idea.

clarfonthey, 2021-07-25:
So, vibrating no longer is infinite until I unplug, and it doesn't crash the driver, but at least it seems that the vibration still goes on for way long (~5ish seconds) and intensity isn't set. I was having issues on Proton earlier so this could be an extension of that, but at least, that's what I'm getting right now.

DanielOgorchock, 2021-07-25:
Do you notice a difference in behavior when switching between usb vs. bluetooth. I've found that for whatever reason rumble is more likely to "miss" commands when the controller is connected via usb. I thought the latest diffs kind of worked around that by repeatedly sending zero-amplitude packets to avoid the several second automatic rumble fade off.

clarfonthey, 2021-07-25:
I'm connected via USB right now; I can try via bluetooth though.

clarfonthey, 2022-02-07:
I'm closing this since I've been using the drivers recently without issues, and there hasn't been any discussion on this since.

closed 2022-02-07 by clarfonthey

sersorrel, 2022-02-25:
fwiw, I just started using a pro controller to play Celeste, and every rumble goes on for a good two or three seconds. this happens both via USB and Bluetooth.

sersorrel, 2022-02-25:
...on the other hand, rumble seems to work absolutely fine in Grapple Dog (via Bluetooth, haven't tested USB).

SeongGino, 2022-04-02:
Hate to necro, but vibration is definitely sustaining way too long when using Virtual Controller mode. Normal pairing feels like it should (but obviously doesn't work with Steam and does not have proper mappings [shameless plug to #85]). Using both Bluetooth/USB.

bernharl, 2022-04-03:
> Hate to necro, but vibration is definitely sustaining way too long when using Virtual Controller mode. Normal pairing feels like it should (but obviously doesn't work with Steam and does not have proper mappings [shameless plug to #85]). Using both Bluetooth/USB.
> 
> EDIT: And between two separate installations of Arch, Kernels 5.15 w/ dkms, and 5.16/5.17 with in-kernel module.

How do you even get rumble to work in virtual controller mode?

SeongGino, 2022-04-03:
> How do you even get rumble to work in virtual controller mode?

It just werked? :woman_shrugging: 
But the manic vibration isn't very pleasant.

clarfonthey, 2022-04-04:
FWIW I think that these different discussions should be split into their own issues; I closed this issue since it was mostly specific to my setup and I couldn't really describe it both before and after it fixed itself, but in your cases it might be, good to help identify your own specific setups so they can be fixed properly.