DanielOgorchock/joycond #71: Joycond does not find joycons

ettmetal, 2021-07-25:
I don't know how else to explain the issue. When connecting a joycon (L or R), bluetooth settings report that it is connected but instead of all lights blinking as expected with joycond, they continues to ping-pong up and down.

running `sudo journalctl -u joycond --follow` shows many lines of output like
> Jan 1 00:00:00 example joycond[0000]: no match found for /sys//devices/pci0000:00/0000:00:14.0/usb1/1-14/1-14:1.0/bluetooth/hci0/hci0:256/0000:0000:0000.0000/input/input39/event258/device/device/leds/*player*4
(real values redacted)

At first, I thought the "//" was maaaybe a mistake, so I changed phys_ctlr.cpp:35 to not add a slash before `devpath`, but after recompiling I get the same output (with one less slash).

DanielOgorchock, 2021-07-25:
What version of hid-nintendo are you using? Are you using the dkms module or building it into the kernel yourself?

DanielOgorchock, 2021-07-25:
Also I assume you're building joycond including this patch? https://github.com/DanielOgorchock/joycond/commit/0ae9c8fc07260786d86b19ca43525422073fe8a9

I thought that patch was backwards compat with the older hid-nintendo naming scheme for the leds, but maybe I broke it.

ettmetal, 2021-07-26:
I'm using dkms, dkms-hid-nintendo v3.1. I didn't see that had been updated, I'll give it a try

ettmetal, 2021-07-26:
Yes, the version of joycond I was building included that patch

closed 2021-07-26 by ettmetal

tarTG, 2021-09-20:
Hello,
i get the exact same issue using the current master-branch of joycond and dkms-hid-nintendo 3.2
i use manjaro-linux with kernel 5.14.
Is there anything i can miss during the installation or should i use an older  #commit to avoid this issue??

tarTG, 2021-09-21:
I found the problem...
dkms-hid-nintendo was not installed, but the package-manager prompted install successful.
I was missing the linux-headers.