DanielOgorchock/joycond #73: Pro Controller doesn't register

Plamper, 2021-08-08:
When connecting my Pro Controller Pressing L+R or pressing + and - does nothing. The controller appears to be still in pairing mode. The picks the controller up just fine. I am on Manjaro using the 5.13.5-1 Kernel. I installed this via the AUR package joycond-git which should just run the install instructions from the readme. When I  run `systemctl status joycond` I get this output:
```
joycond.service - joycond
     Loaded: loaded (/etc/systemd/system/joycond.service; enabled; vendor preset: disabled)
     Active: active (running) since Sun 2021-08-08 15:12:04 CEST; 1min 13s ago
   Main PID: 18466 (joycond)
      Tasks: 1 (limit: 19123)
     Memory: 404.0K
        CPU: 32ms
     CGroup: /system.slice/joycond.service
             └─18466 /usr/bin/joycond

[18466]: no match found for /sys//devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-10/1-10:1.0/bluetooth/>
[18466]: driver_name: Pro Controller
[18466]: MAC: b8:8a:ec:f0:9a:b3
[18466]: adding epoll_subscriber: fd=5
```

Running `dsmeg` shows me:
```
[ 3017.605803] input: Pro Controller as /devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-10/1-10:1.0/bluetooth/hci0/hci0:256/0005:057E:2009.0008/input/input32
[ 3017.605885] hid-generic 0005:057E:2009.0008: input,hidraw6: BLUETOOTH HID v0.01 Gamepad [Pro Controller] on 24:ee:9a:1e:68:ff
```
Already tried the solution from issue #63 but it doesn't work for me. Any help would be appreciated.

DanielOgorchock, 2021-08-08:
It looks like you don't have the hid-nintendo kernel driver installed. Try installing the hid-nintendo-dkms aur package.

Plamper, 2021-08-08:
Thanks that was it! I was led to believe that it was already in the kernel since 5.10 by some random on reddit. Thanks for your time!

closed 2021-08-08 by Plamper

marcothms, 2022-01-11:
With Fedora 35 on `5.15.12-200.fc35.x86_64` I'm still encoutering this issue. I have [dkms-hod-nintendo](https://github.com/nicman23/dkms-hid-nintendo) loaded and also tried a reboot. Trying other methods didn't resolve this issue. I'm also 100% sure, that I own a genuine Pro Controller. I'm using a wired USB connection straight to the laptop.
```● joycond.service - joycond
     Loaded: loaded (/etc/systemd/system/joycond.service; enabled; vendor preset: disabled)
     Active: active (running) since Tue 2022-01-11 10:54:57 CET; 8min ago
   Main PID: 1018 (joycond)
      Tasks: 1 (limit: 18879)
     Memory: 820.0K
        CPU: 68ms
     CGroup: /system.slice/joycond.service
             └─1018 /usr/bin/joycond

Jan 11 11:00:07 itomori joycond[1018]: no match found for /sys//devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.2/3-2.2.1/3-2.2.1.1/3-2.2.1.1:1.0/0003:057E:2009.0019/input/input44/event13/device/>
Jan 11 11:00:07 itomori joycond[1018]: no match found for /sys//devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.2/3-2.2.1/3-2.2.1.1/3-2.2.1.1:1.0/0003:057E:2009.0019/input/input44/event13/device/>
```