DanielOgorchock/joycond #74: Cmake . issues

DarthCaesium, 2021-08-11:
When I run cmake . in the command terminal, I get this:

`-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "/home/DarthCaesium/joycond/CMakeFiles/CMakeOutput.log".
See also "/home/DarthCaesium/joycond/CMakeFiles/CMakeError.log".`

I have already run `clone git https://github.com/DanielOgorchock/joycond/`, `cd joycond `,  `sudo apt install libevdev-dev` and `sudo apt install libudev-dev`.

If it helps, I'm running Linux Mint Cinnamon 20.2.

Auster-South-Anemoi, 2021-11-07:
I had this issue on Ubuntu because I was missing g++. The command on Ubuntu to install it was `sudo apt-get install g++`, but on Mint it could be different, so make sure you use the right one for your distro.

HonbraDev, 2022-04-24:
`sudo dnf install g++` worked on Fedora

CorruptedPlazma, 2022-06-13:
After installing g++ I seem to get the error continued:

```
plaz@ubuntu:~/joycond$ cmake .
-- The CXX compiler identification is GNU 11.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
CMake Error at /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:659 (message):
  pkg-config tool not found
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:825 (_pkg_check_modules_internal)
  CMakeLists.txt:10 (pkg_check_modules)

CMake Error at /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:659 (message):
  pkg-config tool not found
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:825 (_pkg_check_modules_internal)
  CMakeLists.txt:11 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "/home/plaz/joycond/CMakeFiles/CMakeOutput.log".
See also "/home/plaz/joycond/CMakeFiles/CMakeError.log".
```
CMakeError.log is filled with a lot of "No such file or directory" errors.

JaronBurtnett, 2022-09-12:
I'm on Zorin OS 16.1 and am experiencing the same issue.