DanielOgorchock/joycond #75: freezing on "adding_epoll_subscriber"

pjalsGit, 2021-08-12:
i can't pair joycons with SL+SR, L+R, etc

```
● joycond.service - joycond
     Loaded: loaded (/etc/systemd/system/joycond.service; enabled; vendor preset: disabled)
     Active: active (running) since Thu 2021-08-12 12:25:15 EEST; 13s ago
   Main PID: 4821 (joycond)
      Tasks: 1 (limit: 19023)
     Memory: 384.0K
        CPU: 3ms
     CGroup: /system.slice/joycond.service
             └─4821 /usr/bin/joycond

Aug 12 12:25:15 fedora systemd[1]: Started joycond.
Aug 12 12:25:15 fedora joycond[4821]: adding epoll_subscriber: fd=4
```

theres the logs, (yes they are full logs)

pjalsGit, 2021-08-12:
extra info: im on Fedora 34 and i built joycond and dkms-hid-nintendo from source

pjalsGit, 2021-09-01:
is anyone gonna answer?

pjalsGit, 2021-09-24:
> is anyone gonna answer?

githubwasamistake, 2021-12-03:
ok, I got the pad working when I switched usb ports, I previously connected it through a usb hub, everything appears to work except the gyro... does this driver support the gyro?

pjalsGit, 2021-12-03:
what usb port? as far as i know joycons dont have usb, and this issue is:
> i can't pair joycons

githubwasamistake, 2021-12-04:
I was using a pro controller by usb through a usb hub and it was giving the same error as you got
when I decided to plug it directly into a motherboard port, it was finally able to pair properly
there's another joycon driver on github and it appears to have gyro, I'll be trying that out since I wanna try gyro controls in fps

charJe, 2022-08-24:
I have this same problem. I'm using Guix and have tried the package from guix and building from source. Same issue.

infirit, 2022-08-24:
afaik you need to press a couple buttons before things start to work, https://github.com/DanielOgorchock/joycond#usage

charJe, 2022-08-24:
This is probably the problem then. My joycons are only in pairing mode (blinking lights) before connecting to bluetooth. As soon as they connect with bluetooth, one gets a single light, and the other gets two lights.

infirit, 2022-08-25:
I agree the message about polling isn't very helpful. Maybe it can print out the button combinations and mention it's waiting for them untill it does something.

charJe, 2022-08-25:
Maybe I should create a new issue for this.

charJe, 2022-08-28:
I tested this on a different operating system (ubuntu) (same machine). Everything is built from source and using dkms hid-nintendo. I tried this with several configurations. Here is the result of each.

#### configuration 1: no hid-nintendo, no joycond.
Joycons connect with bluetooth. After connecting, the lights continue to bounce back and forth. This seems all normal, obviously we can't move on because we have no driver.

#### configuration 2: hid-nintendo, no joycond.
Joycons connect with bluetooth. After connecting, one joycon becomes single player (1 light) the other becomes second player (2 lights).  Interestingly this is the same thing that happens on my guix computer whether running joycond or not.

#### configuration 3: hid-nintendo, joycond (run with `sudo joycond`)
Joycons connect with bluetooth. After connecting, all the lights go out (different than what is described in the "Usage" document). At this point, joycond has a lot of output whereas before it was just adding `epoll_subscriber`. pressing buttons on the joycons does not make them light up. Pressing buttons on the joycons does make joycond identify the joycon with a log. The joycons can be paired together with the shoulder buttons as intended. A single joycon can go into horizontal mode by using the side buttons (SL & SR); however, only one controller can be in horizontal mode, and the other one needs a partner while the side buttons do nothing (this is most certainly a separate issue though). 

#### conclusion
This indicates that the problem does not lie with my bluetooth configuration or the hid-nintendo, but indeed somehow with joycond not picking up on the joycons. I'm not sure how to continue from here though.

mcs615-2, 2024-11-16:
same, but ZorinOS 17.2 Core and sudo doesn't turn them off, their keep bouncing. before now i used AntimicroX and to set the right profile so that the screenshot button also can be used is hard. Now I left with a not working joycond!