DanielOgorchock/joycond #76: Incorrect pathing for device

alkaris2, 2021-08-29:
Joycond has incorrect pathing for the device;

```
Aug 29 20:52:23 tohru joycond[1173423]: no match found for /sys//devices/pci0000:00/0000:00:14.0/usb1/1-13/1-13:1.0/0003:057E:2009.000C/input/input34>
Aug 29 20:52:23 tohru joycond[1173423]: no match found for /sys//devices/pci0000:00/0000:00:14.0/usb1/1-13/1-13:1.0/0003:057E:2009.000C/input/input34>
Aug 29 20:52:23 tohru joycond[1173423]: no match found for /sys//devices/pci0000:00/0000:00:14.0/usb1/1-13/1-13:1.0/0003:057E:2009.000C/input/input34>
Aug 29 20:52:23 tohru joycond[1173423]: no match found for /sys//devices/pci0000:00/0000:00:14.0/usb1/1-13/1-13:1.0/0003:057E:2009.000C/input/input34>
Aug 29 20:52:23 tohru joycond[1173423]: no match found for /sys//devices/pci0000:00/0000:00:14.0/usb1/1-13/1-13:1.0/0003:057E:2009.000C/input/input34>
Aug 29 20:52:23 tohru joycond[1173423]: no match found for /sys//devices/pci0000:00/0000:00:14.0/usb1/1-13/1-13:1.0/0003:057E:2009.000C/input/input34>
Aug 29 20:52:23 tohru joycond[1173423]: no match found for /sys//devices/pci0000:00/0000:00:14.0/usb1/1-13/1-13:1.0/0003:057E:2009.000C/input/input34>
Aug 29 20:52:23 tohru joycond[1173423]: driver_name: Nintendo Co., Ltd. Pro Controller
Aug 29 20:52:23 tohru joycond[1173423]: MAC: 000000000001
Aug 29 20:52:23 tohru joycond[1173423]: adding epoll_subscriber: fd=5
```

Why is there a double `/`  in `/sys//devices` ?? This shouldn't be like that.  How do I edit this to correct the path?

alkaris2, 2021-09-19:
Anybody?

marcothms, 2022-01-11:
I'm also encountering the same issue.

willf668, 2022-05-23:
Also having this issue.

willf668, 2022-05-23:
I changed the source to replace "/sys/" with "/sys" in the devpath and recompiled, but the result was the same. This seems to be unrelated to the double slash.

infirit, 2022-06-16:
Linux only cares about multiple slashes if you start a path with more than one. Otherwise they will be treated as one.