DanielOgorchock/joycond #85: Switch controller buttons are mismatched

SeongGino, 2021-10-31:
Distro: EndeavorOS | Arch-based
Kernel: linux-xanmod 5.14.15
Installed `hid-nintendo-dkms` and `joycond-git` through AUR

Tested with combined Joycons, a first party Switch Pro Controller over Bluetooth and USB, and an 8bitdo SN30 Pro connected via USB (is not detected by joycond over BT).

In connecting controllers to the host, every button is mapped appropriately except for _two,_ which seems to depend on input method.
 - When dual Joycons, ProCon, and SN30 Pro are paired as a normal Switch controller w/ ZL&ZR, the bottom-most and right-most buttons (Switch-B/Switch-A) are mapped as Xbox B & Xbox A, while the left-most and top-most buttons (Switch-Y/Switch-X) are Xbox X & Xbox Y, **respectively.**
 - When the Pro Controller and SN30 Pro are paired as a Virtual Switch controller for Steam w/ Minus&Plus, the left-most and top-most buttons (Switch-Y/Switch-X) are mapped to Xbox-Y & Xbox-X, while the bottom-most and right-most buttons (Switch-B/Switch-A) are Xbox-A & Xbox-B, **respectively.**

This effects every game in Linux, but especially in the case of using as a Steam-compatible gamepad, this makes playing games confusing or unintuitive when two of the four face buttons are erroneously swapped with respect to a typical Xinput controller. I'm only able to mitigate this in Steam by remapping the suspect inputs back to what their physical orientation should be in Steam's controller configuration, and the Yuzu Switch emulator requires a manual remapping from the autoconfig for the Switch/emulated Pro Controller.

Undeemiss, 2022-05-06:
I am experiencing this issue as well. Are there plans to make any updates to joycond? I find it frustrating that the controllers are "supported" but then clearly and obviously are not working as intended. What is the point of having a compatibility tool if the tool messes up controller bindings?

SeongGino, 2022-05-13:
To be fair, as I'm realizing, the issue is more a `nintendo-hid` (so in-kernel module) issue than a joycond one. But it seems very inconsistent why *my* controller only has one pair (A/B) of buttons reversed but not the other (X/Y).

Undeemiss, 2022-05-14:
Is it? I have this issue if and only if I have joycond installed, as far as I can tell.

infirit, 2022-05-14:
The evdev driver does not expose buttons as X,B,A,Y. It's South, North, East and West. You can use evtest to see if these are correct for you.

Which programs are causing issues (they most likely don't even use the kernel driver).

infirit, 2022-05-14:
Yuzu for example does not use the kernel driver. And yes it get's it wrong, complain to the yuzu folks about that.

Undeemiss, 2022-05-14:
In my use case, it was primarily steam I was having the problem with. I would use this program to make it register as an xbox/generic controller, and then A and B would register like an xbox controller, while X and Y register like a nintendo controller. Either way would be fine if it was consistent, but having it split the controller like that is really awkward.

infirit, 2022-05-16:
The combined joycon device exposes the proper direction, X-> North, Y-> West, A -> East, B-> South. You will need to this in Steam. I think you can define your layout somewhere.

Tropicao, 2022-06-18:
I bumped into the same issue. I use mainline hid_nintendo on kernel 5.18.5 + joycond pulled directly from git, and a Pro controller as virtual controller (not detected by Steam at all if I use "normal" pairing). A and B are inverted but not X and Y. So I tweaked the controller layout directly in Steam to force-invert A and B, as @infirit suggested, I have been able to play with valid layout since then

SeongGino, 2022-06-18:
@Tropicao While that's valid for Steam, it's not for basically anything else - or Wine instances outside of Steam.

My best bet has been using sc-controller, though that lacks rumble support at least on Pro Controllers (and in fairness, Steam's emulated controller does as well). But by that point, I may as well disable the joycond service; the only other reason I'd have it installed is because the `joycond-cemuhook-git` package from AUR has `joycond` as a hard dependency, even though it doesn't need it running, weirdly enough.

leopardheart982, 2023-07-07:
Late on this by a year, but for anyone like me hunting for a solution:

Get evtest and check your inputs to see if they're correct, and then grab AntiMicroX or a similar remapping program. This solves the issue outside of just Steam, assuming it's an issue with the controller itself and not on Steam's end, like it was for me.

SeongGino, 2023-07-08:
Yeeeeah I'm not using more external software for a fix.

It seems that upstream insists on maintaining this behavior (even if it's ergonomically incorrect, which is more important imo), so the solution I landed on is provided by SDL: 
`SDL_GAMECONTROLLER_USE_BUTTON_LABELS=0`
This env var will un-switch the switched buttons so now it behaves as expected. Since most everything that interfaces with or uses the Switch controller--or most any game controller--either provides manual mapping or goes through SDL on Linux anyways, this does effectively solve the issue.
___
On a side note, since Wine and Proton 8.0+ now recognizes a Nintendo Switch Pro Controller as an Xinput controller (even without Steam Input for Switch controllers), I think this makes the Virtual Controller workaround for Pro Controllers, at least where Steam is concerned, a bit moot now. I'm not sure if there will be any changes regarding this in the future, but just something I wanted to note while making a post on this topic of "using Switch controllers on Linux".