DanielOgorchock/joycond #89: movement controls issue on a switch pro controller

sigzegv, 2021-11-18:
I am new to the switch pro controller world on linux. I have the hid_nintendo driver working perfectly when loaded alone without joycond, all the controls and the gyro works fine (eg with yuzu everything fine at the same time).

But I also want to use this config with steam, and I saw the joycond service provide a fix to allow hid_nintendo to not conflict with steam driver.
The issue I have when I start hid_nintendo + joycond are the gyro controls that seems to be detected as a secondary controller. It's not usually an issue for steam games, but I cannot use gyro controls on yuzu anymore after that.

Is this a know limitation or is there a fix ?

infirit, 2021-12-13:
Yuzu and ryujinx use the raw input devices through sdl2 gamepad not the hid_nintendo driver. The secondary evdev device is how hid_nintendo (and playstation) expose the gyro and accel, this is how it should be.

I'll check tonight if I can reproduce this but last time yuzu worked fine with joycond running.

sigzegv, 2021-12-13:
I checked using the pad on yuzu without the hid_nintendo driver works but it's very unconfortable because the 4 little bottom ligths are constantly sweeping (like it doesnt find a slot), (and I don't get the motion sensor detected either without hid_nintendo).
I'll wait for your test for gyro/accel on yuzu with joycond.

infirit, 2021-12-13:
Yeah, the joycond udev rules break yuzu. If the virtual device is supposed to allow for this then it's not working.

sigzegv, 2021-12-16:
is there anything you can do ?