DanielOgorchock/joycond #99: Not detected in parsec but works with gamepad tester

Fourgees, 2022-02-27:
The devices pair successfully (although they blink until I goto https://gamepad-tester.com/) Once there, the joycons works exactly like you'd want them to and have all the inputs correctly mapped. However, when I use parsec, it doesn't detect the controllers. I am running ubuntu 20.04 LTS.

Fourgees, 2022-02-27:
I got it to detect it in parsec by installing this (I guess it needs a new version?) https://github.com/nicman23/dkms-hid-nintendo but the d-pad is not working the L-Joycon in parsec.

klemmm, 2025-01-25:
> I got it to detect it in parsec by installing this (I guess it needs a new version?) https://github.com/nicman23/dkms-hid-nintendo but the d-pad is not working on the L-Joycon in parsec.

Hi! I found this thread with Google, I had the same problem.

Turns out, it appears that parsec doesn't handle correctly the DPAD_UP/DOWN/LEFT/RIGHT sent by joycond, but prefers EV_ABS ABS_HAT0X/HAT0Y events. 

I've managed to make joycond translate  DPAD UP/DOWN/LEFT/RIGHT events to ABS_HAT0X/HAT0Y events, by adding:

#define ANDROID 1

to the file:  virt_ctlr_combined.cpp

and recompiling joycond 

After doing this, parsec recognizes the DPAD