DanielOgorchock/joycond #102: How to install on Steam Deck (SteamOS 3.0)?

n00mkrad, 2022-03-26:
Hello, are there any instructions on how to install and use this on the Steam Deck?

I've tried to get it working but the Arch-based SteamOS 3 does not come with apt or dnf which made it impossible to follow the guide in the readme.

DanielOgorchock, 2022-03-27:
You can probably reference the aur package: https://aur.archlinux.org/packages/joycond-git

n00mkrad, 2022-03-27:
```
yay -Sy joycond
```
```
 -> Could not find all required packages:
        joycond (Target)
```

Someone on reddit told me that `joycond-nicman23-git` is actually the correct package, but I don't get far either:

```
yay -Sy joycond-nicman23-git
```
```
-- Build files have been written to: /home/deck/.cache/yay/joycond-nicman23-git/src/joycond
[ 10%] Building CXX object CMakeFiles/joycond.dir/src/main.cpp.o
/home/deck/.cache/yay/joycond-nicman23-git/src/joycond/src/main.cpp:1:10: fatal error: iostream: No such file or directory
    1 | #include <iostream>
      |          ^~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/joycond.dir/build.make:76: CMakeFiles/joycond.dir/src/main.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:100: CMakeFiles/joycond.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
error making: joycond-nicman23-git
```

n00mkrad, 2022-03-27:
After installing gcc I have a new error message...

`/usr/include/c++/11.1.0/x86_64-pc-linux-gnu/bits/os_defines.h:39:10: fatal error: features.h: No such file or directory`

n00mkrad, 2022-05-02:
So?

Are we just SOL? Or is there a way to get it running on SteamOS?

n00mkrad, 2022-05-17:
> @n00mkrad
> 
> ```
> yay -Sy joycond
> ```
> 
> You have forgotten to append "-git" into package's name, hence the not found output. You must type the full name of a package always! Moreover, just `-S` is good enough.
> 
> ```
> yay -S joycond-git
> ```
> 
> You should do away from this other "nicman23" package, since it's been outdated for two years, and aims for more specific matters and controllers.

`-- Build files have been written to: /home/deck/.cache/yay/joycond-git/src/joycond
[  9%] Building CXX object CMakeFiles/joycond.dir/src/main.cpp.o
In file included from /usr/include/c++/11.1.0/x86_64-pc-linux-gnu/bits/c++config.h:571,
                 from /usr/include/c++/11.1.0/iostream:38,
                 from /home/deck/.cache/yay/joycond-git/src/joycond/src/main.cpp:1:
/usr/include/c++/11.1.0/x86_64-pc-linux-gnu/bits/os_defines.h:39:10: fatal error: features.h: No such file or directory
   39 | #include <features.h>
      |          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/joycond.dir/build.make:76: CMakeFiles/joycond.dir/src/main.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:100: CMakeFiles/joycond.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...`

n00mkrad, 2022-05-18:
> > Yeah no.
> 
> Odd... it installs just flawlessly in my pure Arch Linux distro (my computer that is, I don't have a Steam Deck). Make sure you have got `libevdev`. If so, reinstall it. Could it be your Linux kernel...? For 5.15 ones and before, it's a must to manually install the install `hid-nintendo-dkms` before everything else, either through AUR or its respective Github page.
> 
> And did you recall yourself to input `sudo steamos-readonly disable`? It's for disabling altogether the shields lifted up impending you from accessing deeper the system, allowing writing up into their read-only allowed OS.
> 
> If nothing else, this is likely on SteamOS's end, something of matter on its internal changed core system. Query the issue to Valve, if you haven't already. This link should hopefully aid you on. https://steamcommunity.com/app/1675200/discussions/0/3186864498792143401/

Ah yeah SteamOS uses an older kernel it seems: `5.13.0-valve14-1-neptune-02195-g5b0f749d00fa`. Will report back

n00mkrad, 2022-05-18:
Installing `hid-nintendo-dkms` shows some errors but they are in grey, not red, so idk if it worked or not?

```
(1/2) Arming ConditionNeedsUpdate...
(2/2) Install DKMS modules
==> ERROR: Missing home kernel headers for module hid-nintendo/3.2.
==> ERROR: Missing sys kernel headers for module hid-nintendo/3.2.
==> ERROR: Missing srv kernel headers for module hid-nintendo/3.2.
==> ERROR: Missing opt kernel headers for module hid-nintendo/3.2.
==> ERROR: Missing tmp kernel headers for module hid-nintendo/3.2.
==> ERROR: Missing efi kernel headers for module hid-nintendo/3.2.
==> ERROR: Missing usr kernel headers for module hid-nintendo/3.2.
==> ERROR: Missing run kernel headers for module hid-nintendo/3.2.
==> ERROR: Missing sbin kernel headers for module hid-nintendo/3.2.
==> ERROR: Missing var kernel headers for module hid-nintendo/3.2.
==> ERROR: Missing esp kernel headers for module hid-nintendo/3.2.
==> ERROR: Missing gpu-trace.log kernel headers for module hid-nintendo/3.2.
==> ERROR: Missing lib64 kernel headers for module hid-nintendo/3.2.
==> ERROR: Missing dev kernel headers for module hid-nintendo/3.2.
==> ERROR: Missing bin kernel headers for module hid-nintendo/3.2.
==> ERROR: Missing boot kernel headers for module hid-nintendo/3.2.
==> ERROR: Missing etc kernel headers for module hid-nintendo/3.2.
==> ERROR: Missing lib kernel headers for module hid-nintendo/3.2.
==> ERROR: Missing root kernel headers for module hid-nintendo/3.2.
==> ERROR: Missing mnt kernel headers for module hid-nintendo/3.2.
==> ERROR: Missing proc kernel headers for module hid-nintendo/3.2.
```

n00mkrad, 2022-05-18:
Nah didn't seem to work, I still get `/usr/include/c++/11.1.0/x86_64-pc-linux-gnu/bits/os_defines.h:39:10: fatal error: features.h: No such file or directory`.

Svintooo, 2022-06-24:
So, I had a go myself to get joycons working.
Getting hid-nintendo installed was successful by running the following:

```
sudo steamos-readonly disable
sudo pacman-key --init
sudo pacman-key --populate archlinux
sudo pacman -S bc  #This may not be necessary
sudo pacman -S base-devel
sudo pacman -S linux-neptune-headers
yay -S hid-nintendo-dkms
```

But I get the same error when compiling joycon-git. Not sure how totroubleshoot it.

```
(deck@svintoo-steamdeck ~)$ yay -S joycond-git
:: Checking for conflicts...
:: Checking for inner conflicts...
[Repo Make:4]  jsoncpp-1.9.5-1  libuv-1.43.0-1  rhash-1.4.2-1  cmake-3.22.2-1
[Aur:1]  joycond-git-v0.1.0.r51.gf9a6691-1

==> Remove make dependencies after install? [y/N] y
:: Downloaded PKGBUILD (1/1): joycond-git
  1 joycond-git                      (Build Files Exist)
==> Diffs to show?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> n
:: (1/1) Parsing SRCINFO: joycond-git
[sudo] password for deck: 
Sorry, try again.
[sudo] password for deck: 
resolving dependencies...
looking for conflicting packages...

Packages (4) cmake-3.22.2-1  jsoncpp-1.9.5-1  libuv-1.43.0-1  rhash-1.4.2-1

Total Download Size:   10.39 MiB
Total Installed Size:  63.92 MiB

:: Proceed with installation? [Y/n] 
:: Retrieving packages...
 libuv-1.43.0-1-x86_64                                            226.6 KiB   235 KiB/s 00:01 [#######################################################] 100%
 rhash-1.4.2-1-x86_64                                             164.8 KiB   171 KiB/s 00:01 [#######################################################] 100%
 jsoncpp-1.9.5-1-x86_64                                            94.1 KiB  96.7 KiB/s 00:01 [#######################################################] 100%
 cmake-3.22.2-1-x86_64                                              9.9 MiB  6.98 MiB/s 00:01 [#######################################################] 100%
 Total (4/4)                                                       10.4 MiB  6.78 MiB/s 00:02 [#######################################################] 100%
(4/4) checking keys in keyring                                                                [#######################################################] 100%
(4/4) checking package integrity                                                              [#######################################################] 100%
(4/4) loading package files                                                                   [#######################################################] 100%
(4/4) checking for file conflicts                                                             [#######################################################] 100%
(4/4) checking available disk space                                                           [#######################################################] 100%
:: Processing package changes...
(1/4) installing jsoncpp                                                                      [#######################################################] 100%
Optional dependencies for jsoncpp
    jsoncpp-doc: documentation
(2/4) installing libuv                                                                        [#######################################################] 100%
(3/4) installing rhash                                                                        [#######################################################] 100%
(4/4) installing cmake                                                                        [#######################################################] 100%
Optional dependencies for cmake
    qt6-base: cmake-gui
:: Running post-transaction hooks...
(1/4) Arming ConditionNeedsUpdate...
(2/4) Updating the MIME type database...
(3/4) Updating icon theme caches...
(4/4) Updating the desktop file MIME type cache...
==> Making package: joycond-git v0.1.0.r51.gf9a6691-1 (Fri 24 Jun 2022 11:31:50 AM CEST)
==> Retrieving sources...
  -> Cloning joycond git repo...
Cloning into bare repository '/home/deck/.cache/yay/joycond-git/joycond'...
[... 56 lines cut ...]
      |          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/joycond.dir/build.make:76: CMakeFiles/joycond.dir/src/main.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:100: CMakeFiles/joycond.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
checking dependencies...

Packages (4) cmake-3.22.2-1  jsoncpp-1.9.5-1  libuv-1.43.0-1  rhash-1.4.2-1

Total Removed Size:  63.92 MiB

:: Do you want to remove these packages? [Y/n] 
:: Processing package changes...
(1/4) removing cmake                                                                          [#######################################################] 100%
(2/4) removing rhash                                                                          [#######################################################] 100%
(3/4) removing libuv                                                                          [#######################################################] 100%
(4/4) removing jsoncpp                                                                        [#######################################################] 100%
:: Running post-transaction hooks...
(1/4) Arming ConditionNeedsUpdate...
(2/4) Updating the MIME type database...
(3/4) Updating icon theme caches...
(4/4) Updating the desktop file MIME type cache...
```

brumeire, 2022-07-21:
After a few hours troubleshooting this, I could make it work. 

Apparently the steam deck pre-installed packages are missing files, which creates problems such as this one. (https://www.reddit.com/r/SteamDeck/comments/t92ozw/for_compiling_c_code/)

So I ran the command ` pacman -S glibc linux-api-headers` suggested in the reddit thread, the build could go further but failed because of another missing file from an other package. I didn't bother to troubleshoot which one it was and instead re-installed all the packages with missing files with the following command and the build worked fine this time.
`sudo pacman -Qknq | cut -d' ' -f 1 | sudo pacman -S -`

I'm not sure if valve deleted those files just for disk space or another reason and you might need to re-do this after each steam deck update but so far it works for me

RogierJulien, 2022-07-28:
Cant get it to work either, tried now everything, any of you fixed this? I am about to give up trying to make it work to connect my joycons as a single device. 
Bluetooth connect works for both joycons, but no input is recognized. ..

nupanick, 2022-07-29:
> After a few hours troubleshooting this, I could make it work.
>
> Apparently the steam deck pre-installed packages are missing files, which creates problems such as this one. [...]

This worked for me! I went through the missing packages one at a time instead of using your one-liner, but I still got it working after a little tinkering.
First, run `yay -S joycond-git`. See what file is missing.
Then, run `pacman -Qk 2>/dev/null | grep udev`, replacing 'udev' with whatever comes after the 'lib' part of the missing file.
Then, run `sudo pacman -S udev`, this time replacing 'udev' with the name of the missing package you found previously.
Repeat these steps until the install is successful.

iamra, 2022-08-18:
Thanks to everyone for getting this to work.
I was able to get everything installed from a default state with the following commands as mentioned before:

`sudo steamos-readonly disable
sudo pacman-key --init
sudo pacman-key --populate archlinux
sudo pacman -S bc  #This may not be necessary
sudo pacman -S --overwrite \* base-devel
sudo pacman -S linux-neptune-headers
sudo pacman -S glibc linux-api-headers
sudo pacman -Qknq | cut -d' ' -f 1 | sudo pacman -S -

yay -S hid-nintendo-dkms
yay -Sy joycond-git
sudo systemctl enable --now joycond
`
Observations:
Combined Joycons work as expected and picked up by Steam as generic controller that can be configured.
Single Joycons work as expected, however, in Steam, for some reason, Only the IMU version of each controller is listed to be configured, these only contain the gyro axis.
One final step to get all this to work properly at this point is to get Steam to show the non IMU single joycons instead.