DanielOgorchock/joycond #104: Rumble on Pro Controller is way too strong

FridgeRacer, 2022-05-26:
I'm using an official Nintendo Switch Pro Controller on Arch Linux with the Linux-Zen kernel.
The rumble across all games is way too strong. Is there any way I, or with a patch, can reduce the maximum (or overall) rumble strength?

As it is right now, not only is the controller unbearably noisy, it also makes my hands and wrists hurt (which some games on Switch do as well, if they haven't tuned the rumble properly!).
In many games I can turn off rumble and not mind, but for some games it'd be nice to have a little bit of rumble feedback.

Games tested if it matters:
- Final Fantasy XI (Wine)
- Re-Volt (Native)
- Tetris Effect (Wine)
- Psychonauts 2 (Proton)

Apologies in advance if this isn't the right place to post this, or if I missed anything.

SeongGino, 2022-06-04:
Ditto. This makes Pro Controllers virtually unusable imo.

sersorrel, 2022-07-05:
there are some similar reports at the end of #69, though they sound like they're to do with the length of vibration rather than the strength...

acerspyro, 2023-06-15:
Is this issue still valid?

dweipert-3138720606, 2024-08-08:
Could maybe make it configurable in max/min strength or being able to turn rumble on/off altogether.

Apteryks, 2024-09-29:
I don't know how they do it, but using RetroArch, you can reduce the haptic force feedback from 100% to something lower (and yes I agree, the default 100% appears too strong -- the controller sounds like it's pushed to the limit).

In RetroArch, you go in `Settings -> Input -> Haptic Feedback/Vibration -> Vibration Strength`

I set it to 5% and I like it better.

BeeFox-sys, 2025-03-22:
I'd love an option to configure the rumble strength too, or disable it entirely

tomben13, 2025-08-25:
This is probably a bug in the hid_nintendo driver. Although I'm not using joycond, I had the same issue with games using the evdev interface (which gets its data from hid_nintendo). I made this patch to the kernel which works for me.
Try applying the patch and compiling the hid_nintendo module to see if it fixes the issue.

```
From 6b5d0b5d77d4c624757ff09d878d8aa68b7556be Mon Sep 17 00:00:00 2001
From: Tom Benham <tom.benham13@gmail.com>
Date: Mon, 25 Aug 2025 15:39:59 +0200
Subject: [PATCH 1/3] HID: nintendo: fix way too strong rumble

---
 drivers/hid/hid-nintendo.c | 34 +++++++++++++++++++---------------
 1 file changed, 19 insertions(+), 15 deletions(-)

diff --git a/drivers/hid/hid-nintendo.c b/drivers/hid/hid-nintendo.c
index fb4985988615..8a1f92600750 100644
--- a/drivers/hid/hid-nintendo.c
+++ b/drivers/hid/hid-nintendo.c
@@ -297,8 +297,8 @@ static const struct joycon_rumble_amp_data joycon_rumble_amplitudes[] = {
 	{ 0xc2, 0x8070,  940 }, { 0xc4, 0x0071,  960 }, { 0xc6, 0x8071,  981 },
 	{ 0xc8, 0x0072, joycon_max_rumble_amp }
 };
-static const u16 JC_RUMBLE_DFLT_LOW_FREQ = 160;
-static const u16 JC_RUMBLE_DFLT_HIGH_FREQ = 320;
+static const u16 JC_RUMBLE_DFLT_LOW_FREQ = 140;
+static const u16 JC_RUMBLE_DFLT_HIGH_FREQ = 180;
 static const unsigned short JC_RUMBLE_ZERO_AMP_PKT_CNT = 5;
 #endif /* IS_ENABLED(CONFIG_NINTENDO_FF) */
 static const u16 JC_RUMBLE_PERIOD_MS = 50;
@@ -1865,20 +1865,22 @@ static struct joycon_rumble_amp_data joycon_find_rumble_amp(u16 amp)
 	return data[i];
 }
 
-static void joycon_encode_rumble(u8 *data, u16 freq_low, u16 freq_high, u16 amp)
+static void joycon_encode_rumble(u8 *data, u16 freq_low, u16 freq_high, u16 amp_low, u16 amp_high)
 {
 	struct joycon_rumble_freq_data freq_data_low;
 	struct joycon_rumble_freq_data freq_data_high;
-	struct joycon_rumble_amp_data amp_data;
+	struct joycon_rumble_amp_data amp_data_low;
+	struct joycon_rumble_amp_data amp_data_high;
 
 	freq_data_low = joycon_find_rumble_freq(freq_low);
 	freq_data_high = joycon_find_rumble_freq(freq_high);
-	amp_data = joycon_find_rumble_amp(amp);
+	amp_data_low = joycon_find_rumble_amp(amp_low);
+	amp_data_high = joycon_find_rumble_amp(amp_high);
 
 	data[0] = (freq_data_high.high >> 8) & 0xFF;
-	data[1] = (freq_data_high.high & 0xFF) + amp_data.high;
-	data[2] = freq_data_low.low + ((amp_data.low >> 8) & 0xFF);
-	data[3] = amp_data.low & 0xFF;
+	data[1] = (freq_data_high.high & 0xFF) + amp_data_high.high;
+	data[2] = freq_data_low.low + ((amp_data_low.low >> 8) & 0xFF);
+	data[3] = amp_data_low.low & 0xFF;
 }
 
 static const u16 JOYCON_MAX_RUMBLE_HIGH_FREQ	= 1253;
@@ -1906,11 +1908,12 @@ static void joycon_clamp_rumble_freqs(struct joycon_ctlr *ctlr)
 	spin_unlock_irqrestore(&ctlr->lock, flags);
 }
 
-static int joycon_set_rumble(struct joycon_ctlr *ctlr, u16 amp_r, u16 amp_l,
+static int joycon_set_rumble(struct joycon_ctlr *ctlr, u16 amp_weak, u16 amp_strong,
 			     bool schedule_now)
 {
 	u8 data[JC_RUMBLE_DATA_SIZE];
-	u16 amp;
+	u16 amp_low;
+	u16 amp_high;
 	u16 freq_r_low;
 	u16 freq_r_high;
 	u16 freq_l_low;
@@ -1924,17 +1927,18 @@ static int joycon_set_rumble(struct joycon_ctlr *ctlr, u16 amp_r, u16 amp_l,
 	freq_l_low = ctlr->rumble_ll_freq;
 	freq_l_high = ctlr->rumble_lh_freq;
 	/* limit number of silent rumble packets to reduce traffic */
-	if (amp_l != 0 || amp_r != 0)
+	if (amp_weak != 0 || amp_strong != 0)
 		ctlr->rumble_zero_countdown = JC_RUMBLE_ZERO_AMP_PKT_CNT;
 	spin_unlock_irqrestore(&ctlr->lock, flags);
 
+	amp_low = amp_strong * (u32)joycon_max_rumble_amp / 65535;
+	/* For high frequency, divide amp by 8 to simulate light rumble motor */
+	amp_high = (amp_weak >> 3) * (u32)joycon_max_rumble_amp / 65535;
 	/* right joy-con */
-	amp = amp_r * (u32)joycon_max_rumble_amp / 65535;
-	joycon_encode_rumble(data + 4, freq_r_low, freq_r_high, amp);
+	joycon_encode_rumble(data + 4, freq_r_low, freq_r_high, amp_low, amp_high);
 
 	/* left joy-con */
-	amp = amp_l * (u32)joycon_max_rumble_amp / 65535;
-	joycon_encode_rumble(data, freq_l_low, freq_l_high, amp);
+	joycon_encode_rumble(data, freq_l_low, freq_l_high, amp_low, amp_high);
 
 	spin_lock_irqsave(&ctlr->lock, flags);
 
-- 
2.50.1

```

keysmashes, 2025-08-28:
That patch changes how rumble feels (it seems like it now vibrates at a lower frequency, maybe?) but rumble still lasts for way too long.

keysmashes, 2025-09-14:
It's maybe a bit weaker, I think?