DanielOgorchock/joycond #107: Not able to connect joycons

Cheeseboy8020, 2022-06-25:
I am on kubuntu 22.04. I am able to connect to my joycons with bluetooth but the LED keeps going up and down instead of blinking all four LEDs. When I disconnect the joycons in bluetooth the LED turns off. I installed both joycond and hid-nintendo.

applecuckoo, 2023-02-25:
Hi there @Cheeseboy8020 and @klarkc (you reacted to the post). Not sure if you're still having this problem? If so, then try installing `libudev-devel` then rebuild and reinstall. Solved the problem for me.

Robro0-625, 2024-06-08:
I am experiencing the same problem and your suggestion hasn't helped, however when i launch Joycond i can see the controllers are detected, they just aren't paired properly on button press.

Aadit420, 2025-02-20:
I am also facing the same issue in Pop OS. Any solution so far?