DanielOgorchock/joycond #109: gyro not working (secondary controller is not updating)

Xephobia, 2022-07-29:
Hi, it seems that the gyro isnt working on my pro controller, I'm using the virtual mode for steam and the secondary controller representing the gyro is here, however, it's not updating.

banaanihillo, 2023-01-29:
Could you elaborate on what you mean by "the secondary controller representing the gyro is here"?
I can't get gyro to work with Steam on either of my Nintendo Switch controllers, that is, the Pro Controller and the Joy-Cons, and I'm wondering if it's the same issue, and if perhaps you've found a fix for it.

I can see them all working via [jstest](https://jstest-gtk.gitlab.io/) but Steam doesn't detect the Pro Controller at all if I connect it with L + R, and it's detected as a Joy-Con pair if I connect with Minus + Plus. I'm not sure if the latter issue has anything to do with the gyro, all I know is that it appears not to work.

Linwer, 2023-02-01:
i am also having an issue with steam not detecting gyro from my nintendo switch pro virtual (plus minus) controller. peculiarly it is also detected as a pair of joy cons on steam but i'm pretty sure that's deliberate

jclds139, 2023-06-26:
I think this may be due to joycond creating new virtual `event*` devices for the buttons and sticks, but expecting the original IMU devices to work. Steam, etc, can probably only find the IMU device if it's paired directly with the normal input devices. Concrete example, my left joycon is `event17` with `event18` as its IMU, right joycon is `event19` with `event20` as its IMU. The joycond "Combined" device is `event21`, but there's no `event22` with the combined IMU data so it's as though there is no IMU.
If this is indeed the problem, it sounds to me like joycond also needs to do the IMU data combining from both sides and feed it into a second virtual IMU device.
For a start, we could just pick one IMU and pass it through on the virtual device before figuring out the combining logic.

Beethoven-n, 2024-06-18:
can second the issue! i opened the controllers up in a different program, and the individual controllers respond just fine to gyro aim, but the combined pair does not.