DanielOgorchock/joycond #110: Swith pro controller (and sometimes analog sticks) not working in steam

Expressive-Synapse, 2022-09-08:
Distoro: Fedora36
Controller: Official switch pro controller

Note: I am using the virtual mode since steam doesn't recognize the controller otherwise.

As the title says, when trying to use the switch pro controller with steam, the dpad inputs don't get recognized, and in one game the analog stick doesn't get recognized either. I have made sure that the dpad on the controller works by testing it with my switch, and i've confirmed that my computer is reading the dpad inputs by using dolphin, it's just steam that doesn't recognize the dpad inputs for some reason. Any help would be appreciated. Thank you.

fliperama00, 2022-09-18:
On Steam, go to the controller configuration, click on the detected controller, then click on define buttons (? I don't know how it is exactly in the English version, but is on the side of "calibrate"). Just remap the four directions to the actual input of your controller.

Expressive-Synapse, 2022-09-18:
> On Steam, go to the controller configuration, click on the detected controller, then click on define buttons (? I don't know how it is exactly in the English version, but is on the side of "calibrate"). Just remap the four directions to the actual input of your controller.

See at first I though to try that too but whenever I try to define the buttons nothing happens. I can select an input but no matter what button on the controller I press is doesn't log it as the new input.