DanielOgorchock/joycond #121: Switch Pro Controller recognised as device 057e 2008 (Nintendo Switch Joy-Con L/R) in steam rather than 057e 2009 (Nintendo Switch Pro Controller)

edbefee3-3888-462a-9411-741b7e9eb54e, 2023-03-30:


Norgus, 2023-04-04:
OK so I found a KDE tool that lets you see controller input & the "virtual pro controller" device does in fact appear to recognise input from the square button (shows as button 5) and the d-pad (shows as an axis on axes 5 and 6)
[image]
I tried turning it off and using the generic pairing (both triggers) and as far as this KDE widget is concerned the real controller seems to behave exactly like the virtual one. It must be a steam issue, but I have no idea how to fix it.

edbefee3-3888-462a-9411-741b7e9eb54e, 2023-04-05:
the main problem I have with it is that gyro/accelerometer don't work

edbefee3-3888-462a-9411-741b7e9eb54e, 2023-04-05:
BTW it isn't a Steam thing, it seems that it's an SDL thing

dexterdy, 2023-08-06:
The same dpad issue also happens when you pair two joycon controllers as a single controller, btw. This button mapping problem does not happen if you connect the controller without joycond

sonich2401, 2024-03-04:
My pro controller maps as a "Nintendo Switch Left Joy-Con" with and without joycond. None of the input works at all. Do note it is a 3rd party controller but it works fine on a real switch and is detected as a pro controller by my switch.

Cake-Chaser, 2024-05-04:
@sonich2401  how did you fix it. i too  have no gyro on combined joycons

sonich2401, 2024-05-04:
> @sonich2401  how did you fix it. i too  have no gyro on combined joycons
> 
> 

I figured out the issue is with hid_nintendo. I switched to hid_nx. However, joycons seem a bit laggier with hid_nx over hid_nitnendo. Imo hid_nx is 100% worth it. You have to disable hid_nitnendo and then install hid_nx. It is a drop in replacement for hid_nitnendo so everything should work perfectly.

sonich2401, 2024-05-06:
> @sonich2401  how did you fix it. i too  have no gyro on combined joycons
> 
> 

Also I used joycond-cemuhook to get gyro