DanielOgorchock/joycond #126: How can I clone the joycond project?

lebonhomme-mangos, 2023-07-14:
Hi,

I'm trying to make my NS Joystick on my retropie Raspberry Pi Zero W. So I follow the official guide here: https://retropie.org.uk/docs/Nintendo-Switch-Controllers/

However when I type git clone https://github.com/DanielOgorchock/joycond.git, I face to incredible security issues for "just cloning" a repo. Firstly it prompts a username and password, making me to create an account on Github. But when I type it, it tells that "Support for password authentication was removed  on" blablabla [...] redirecting me to https://docs.github.com/en/get-started/getting-started-with-git/about-remote-repositories#cloning-with-https-urls. I'm not used at all with all that Github things, but I created 2 tokens (1 fine-grained and 1 classic) and I tried both instead of my password when I try to clone... And then it simply tells "Repository not found". I'm so confused! Why is it so hard? For information, I didnt know what permissions to put on the tokens so I took them all, is that the problem? If so, what token must I generate and with wich configuration? I am on retropie terminal and I have no GUI for the moment, so impossible to use the Github desktop app...

Thanks in advance for your help,
Best regards,
A random retro player.

sp1ritCS, 2023-07-15:
Doesn't happen to me. I can clone git clone https://github.com/DanielOgorchock/joycond.git without issue. The Authentication thingy / Repository not found should only happen if you try to clone a repository that doesn't exist.
Have you checked for typos in the URL?

Apteryks, 2024-09-24:
This tracker needs some bug triaging.