DanielOgorchock/joycond #128: Pro controller gyro axes mixed up

zeusttu, 2023-10-01:
I managed to pair a Pro Controller, and it'r recognised mostly fine. Motion controls are also being recognised, but every motion is misinterpreted.

When testing from Steam's settings dialog (I read on the internet Steam used to use its own driver but it probably no longer does?):

- When I rotate the controller within the horizontal plane, it gets interpreted as moving the left/right side up/down.
- When I move the left/right side up/down, it gets interpreted as moving the front/back side up/down.
- When I move the front/back side up/down, it gets interpreted as rotating the controller within the horizontal plane.

When testing with `evtest`:

- Rotating the controller within the horizontal plane seems to correspond to the `ABS_RZ` axis.
- Moving the left/right of the controller up/down seems to correspond to the `ABS_RX` axis.
- Moving the front/back of the controller up/down seems to correspond to the `ABS_RY` axis.

I know these axes have official names, but I don't know which is which, so I hope I described each axis in an understandable way. I also don't know whether this is an issue with Joycond or with the underlying `hid-nintendo` driver, so if I created this issue for the wrong project, feel free to redirect me to the correct one.

Could there be a mismatch between Joycond and the underlying driver regarding the gyro axes? I think I read something somewhere about something like that happening in the past.

For what it's worth:

```
$ uname -r
6.5.5-arch1-1

$ pacman -Qi joycond-git
Name            : joycond-git
Version         : v0.1.0.r52.g5b590ec-1
Description     : Userspace daemon to combine joy-cons from the hid-joycon kernel driver
Architecture    : x86_64
URL             : https://github.com/DanielOgorchock/joycond
Licenses        : GPL3
Groups          : None
Provides        : joycond-git
Depends On      : libevdev
Optional Deps   : hid-nintendo-dkms: provides driver for pre-5.16 kernels
Required By     : None
Optional For    : None
Conflicts With  : joycond-git
Replaces        : None
Installed Size  : 120.84 KiB
Packager        : Unknown Packager
Build Date      : Sun 01 Oct 2023 02:29:35 PM CEST
Install Date    : Sun 01 Oct 2023 02:30:02 PM CEST
Install Reason  : Explicitly installed
Install Script  : Yes
Validated By    : None
```

The installed commit hash seems to be `5b590ecc9bca181d8bc21377e752126bc9180319`, which matches the current state of the `master` branch. I don't have `hid-nintendo-dkms` installed because my kernel is recent enough to have it built-in.

zeusttu, 2023-10-01:
Further detail: the controller wasn't fully up-to-date when I created this issue, but it is now, and the issue persists with no change in the behaviour I'm seeing.

Victor-Silva-3000, 2023-12-23:
I'm currently having the same issue. Have you found any workarounds?

edbefee3-3888-462a-9411-741b7e9eb54e, 2024-02-12:
Gyro is fine actually, the accelerometer axes are messed up in the SDL binding (you can see this in dolphin-emu by setting both the gyro and accelerometer), however, Joycond_Cemuhook is correct
so, by cross examining, the correct evdev mapping is: X+ is front, Y+ is left, and Z+ is up

edbefee3-3888-462a-9411-741b7e9eb54e, 2024-02-12:
[image]
it seems to be an SDL thing, this results in correct motion capture

DarBar92, 2024-04-08:
I am also having this issue. Have you or anyone found a workaround?

Victor-Silva-3000, 2024-04-10:
My solution was to just drop joycond altogether and use the built-in driver in the Linux Kernel for the Nintendo Controllers. 

I also had an issue related to the controller not being recognized in any application due to it being assigned (for some reason) to the root user. If anyone has a similar issue feel free to ask and I will try to elaborate as best I can.

Note that at the time I posted this I was using an older version of Manjaro KDE and after updating everything worked flawlessly.