DanielOgorchock/joycond #134: BlueZ security patch breaks Joy-Con connections

applecuckoo, 2023-12-08:
A recently uncovered flaw in the Bluetooth protocol (CVE-2023-45866) has shown that an unauthenticated Bluetooth connection allows the injection of malicious keystrokes. If you want to read up on that flaw, you can click [here](https://github.com/skysafe/reblog/blob/main/cve-2023-45866/README.md) for a write-up from the discoverer. Down the line, it seems to affect the connection of Joy-Cons to the system.

I had to do a bit of tinkering and rolling back to determine that the patch broke the operation with the Joy-Cons. I'm writing here in the hopes that there is a secure workaround (i.e. using Nintendo's pairing system) to this problem. Otherwise, it might spell the end of Joy-Cons on Linux (and potentially other systems).

DanielOgorchock, 2023-12-13:
Oh no! That's a bummer to hear. Yeah I wonder if we can use the usb-based pairing procedure.

applecuckoo, 2023-12-13:
Hello! I can confirm that the Pro Controller still works. Maybe I should start a discussion in the [BlueZ repo](https://github.com/orgs/bluez/discussions)? The maintainer seems to be quite active there.

DanielOgorchock, 2023-12-13:
Oh interesting. So pairing the pro controller directly via bluetooth still works but not the joycons?

applecuckoo, 2023-12-13:
@DanielOgorchock No, neither controllers work as in Bluetooth can't connect to the controllers.

applecuckoo, 2023-12-27:
Welp, I guess we'll learn more on this vulnerability when [ShmooCon](https://www.shmoocon.org/about_shmoocon/) rolls around - the researcher who discovered the attack, Marc Newlin, is doing a talk there on the 13th of January next year in the 'Bring It On' set of talks.

applecuckoo, 2023-12-30:
Seems like the problem will go away if you pair with the controller, disconnect from it and then reconnect to it again. Closing this issue for now. @hajosattila Feel free to post here if you still need more help.

closed 2023-12-30 by applecuckoo

hajosattila, 2023-12-30:
> @hajosattila Feel free to post here if you still need more help.

It works perfectly!