DanielOgorchock/joycond #137: D-Pad not functional

ibanman555, 2023-12-29:
I've installed this on Raspberry PI to replace my old controller and works well except I have zero input on the D pad. Any work around or fixes?

applecuckoo, 2023-12-29:
Hello! This is an interface for Switch controllers to act as an input for games and emulators. I'm not sure this is what you're trying to do?

ibanman555, 2023-12-29:
Sorry for the lack of details. I am using Moonlight on Raspberry PI to stream PC games to a large screen TV, and using Switch Pro Controllers to play the games. I've installed hid-nintendo and your joycond program, and I am able connect and use the controllers as expected to the Pi, however the only buttons on my Switch Pro Controllers not responding are the D-Pad buttons.

applecuckoo, 2023-12-30:
@ibanman555 Prepare yourself, here's a bit of a guide to troubleshooting your situation:

1. Try installing `evemu-record` with `sudo apt install evemu-tools`.
2. Connect your controller to the Pi.
3. Type `evemu-record` into the terminal and hit 'Enter'. You will then be met with a prompt to select an event number.
4. To the left of `Nintendo Switch Pro Controller` there will be something like `/dev/input/eventXX`. The XX represents the number you'll type into the prompt.
5. Enter the number.
6. Start pressing the D-pad. You should see stuff flowing across the screen.
7. Look for `ABS_HAT0X` and `ABS_HAT0Y`. If you see that, that means that the Linux input system can hear your controller just fine, so the problem will be down to mapping.

Hope this helps!

ibanman555, 2023-12-31:
Thanks for the tips @applecuckoo 

Pro Controller input is indeed being seen and registered in `evemu-record`. AntiMicroX is also registering all controller buttons and input. In fact, the controller is fully functional when paired Bluetooth without any controller mapping applications, also. 

In Terminal, I enter `sudo systemctl enable --now joycond`. The Pro Controller will show 4 blinking lights once connected to BT. Here are the results:

If I pair the controller with L&R shoulder buttons, I get zero input from the controller in Moonlight. I do see input in Terminal and AntiMicroX. 

If I pair the controller with + & - buttons on the controller (virtual controller?), I get input from all but the D-Pad. I do see input in Terminal and AntiMicroX, again. 

So I am a little stumped on why enabling joycond is rendering the use of the controller this way, I figured it was a problem that might be an issue for others, or if there is something happening specifically on my end.

applecuckoo, 2023-12-31:
All this sounds like the problem is with the button mapping. The Linux kernel drivers are doing their job correctly, so you may want to close this issue and join Moonlight's [discord server](https://discord.com/invite/CGg5JxN) and try and do some troubleshooting there. Just out of curiosity, I tested my own Pro Controller with a button mapping app utilising the SDL2 API that Moonlight also uses, and the mapping seemed to be correct by default.

closed 2023-12-31 by ibanman555