DanielOgorchock/joycond #138: Joycond cannot find my joycon after bluetooth connected.

DasCrystal, 2024-01-07:
My system is Fedora 39 Workstation (Kernel Version: 6.6.8-200.fc39.x86_64), with conditions below:

- Kernel Module "hid_nx" is installed
[image]
- "joycond" install and enabled
[image]
- Joycon connected
[image]

But joycon didn't vibrate. Couldn't see in the yuzu, either.
(It's works good on windows (on same computer) with BetterJoy, so it shouldn't be a hardware problem.)

And in joycond service, it just stuck at "adding epoll_subscriber: fd=4", no further react.

If you need more information, please tell me.

Beethoven-n, 2024-01-27:
same here on arch, kernel 6.7.1

erudic, 2024-03-03:
Hello,

Are lights on your joycon moving from left to right? If yes, then you might be able to resolve the problem same way I did by following steps from this comment: https://github.com/bluez/bluez/issues/673#issuecomment-1849132576

DasCrystal, 2024-03-20:
First, thanks to @erudic for providing a great approch for this issue.
although that way isn't working for me, but int the same issue page, I found a resolution below:

Set these configs: (sudo vi or something else, and save them)
"UserspaceHID=true", in "/etc/bluetooth/input.conf"
"ClassicBondedOnly=false", also in "/etc/bluetooth/input.conf"
"FastConnectable = true" in "/etc/bluetooth/main.conf"

Then reboot.

And you might found your Joy-Con Works again!
This worked for me.

closed 2024-03-20 by DasCrystal