DanielOgorchock/joycond #142: udev rules make steam fail to recognize controllers

ngraham20, 2024-06-23:
I installed joycond per the instructions, but once it was installed, Steam no longer could see the controllers. This was only resolved when I uninstalled joycond manually, and specifically when I removed the udev rules that joycond creates.

TheRamsWay, 2024-07-01:
Seems to be by design, since in [89-joycond.rules](https://github.com/DanielOgorchock/joycond/blob/master/udev/89-joycond.rules), the first 5 lines are meant to block Steam.
However, this doesn't only break Steam but also SDL's ability to merge the buttons-only and motion-only devices, for example causing Ryujinx to be unable to use motion controls on Pro Controllers. Commenting out line 3 and 5 of that file seems to let both joycond and SDL work as they should.
If I remember correctly, Steam has some settings to disable Steam Input on Nintendo controllers anyway, meaning that joycond shouldn't really need to hide them from it. They could probably just be removed.

ngraham20, 2024-07-01:
> Steam has some settings to disable Steam Input on Nintendo controllers anyway

Why would you want to disable Steam Input though, isn't the fact that Steam can handle any controller a good thing?

sp1ritCS, 2024-07-03:
> Why would you want to disable Steam Input though, isn't the fact that Steam can handle any controller a good thing?

I believe the reasoning is to avoid issues that occur in games that detect both the virtual steam input controller and the one from hid-nintendo/joycond, resulting in multiple connected controllers from the perspective of the game (they often don't let you disable one), even tho both of them are the same physical device.

Depending on what buttons you press during registration to joycond, it either lets steam recognize it or not. But it has issues with the gyro.

pizzapielordtemptest, 2024-07-17:
how do you remove the udev rules that joycond creates?

TheRamsWay, 2024-07-18:
Edit `/usr/lib/udev/rules.d/89-joycond.rules` and comment out the two lines starting with `KERNEL=="hidraw*"` at the beginning of the file. Or delete the file but removing only those two lines seems to be enough, I haven't tested deleting the whole file.