DanielOgorchock/joycond #145: Joycond is not detecting Controller Pro or Joy-Cons after upgrading bluez. Normal function prior bluez 5.64

Mxzcabel, 2024-08-13:
Joycond is unable to detected any controller via bluetooth when updating bluez to currently versions. Plugging Controller Pro in USB works as its intent.

Mxzcabel, 2024-09-08:
For anyone wondering if there's some solution, found out I just needed to edit my ```/etc/bluetooth/input.conf``` with:
```
# Enable HID protocol handling in userspace input profile
# Defaults to true (Use UHID instead of kernel HIDP)
UserspaceHID=true

# Limit HID connections to bonded devices
# The HID Profile does not specify that devices must be bonded, however some
# platforms may want to make sure that input connections only come from bonded
# device connections. Several older mice have been known for not supporting
# pairing/encryption.
# Defaults to true for security.
ClassicBondedOnly=false
```
Those set options worked out and now my controllers are connecting with no struggles.

Apteryks, 2024-09-24:
Works for me without any extra configuration using bluez 5.72 on top of my Guix System running GNOME 44.1.