DanielOgorchock/joycond #146: Doesn't work on Linux Mint22 Kernel 6.8

Arjuna2227, 2024-08-19:
Joy-Cons connect through Bluetooth, but the lights keep blinking and they are not recognized in any type of software. 

Joycond enabled according to sudo systemctl status joycond. 

Button combinations don't work, both for using them on their own and together as one input device. 

Issues is wide-spread on reddit and no known solution has been shared online yet.

axolotl-being, 2024-09-08:
I was experiencing this same issue, only after trying to reconnect the joycons after initially pairing them, it would often crash my system seemingly at random. and then I'd need to do a full powercycle to get the bluetooth adapter working again.

However, I think I've finally found a solution. And that solution is to simply uninstall blueman manager and use a different bluetooth manager instead like bluetuith. So far reconnecting seems to work properly and I haven't experienced any crashes so far.

migas2g, 2025-08-27:
change kerenel to 6.11