DanielOgorchock/joycond #147: Failed to enable unit: Unit file joycond.service does not exist

J-H-Freedman, 2024-09-07:
Off a fresh install, it does not produce joycond.service

```
a@pop-os:~$ sudo rm -r joycond/
a@pop-os:~$ sudo git clone https://github.com/DanielOgorchock/joycond.git
Cloning into 'joycond'...
remote: Enumerating objects: 586, done.
remote: Counting objects: 100% (55/55), done.
remote: Compressing objects: 100% (28/28), done.
remote: Total 586 (delta 31), reused 32 (delta 27), pack-reused 531 (from 1)
Receiving objects: 100% (586/586), 108.45 KiB | 1.04 MiB/s, done.
Resolving deltas: 100% (378/378), done.
a@pop-os:~$ sudo cmake joycond/
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jhfreedman
a@pop-os:~$ sudo systemctl enable --now joycond
Failed to enable unit: Unit file joycond.service does not exist.
a@pop-os:~$ 
```

What should I do to fix that?

J-H-Freedman, 2024-09-07:
forgot 
`sudo make install`

I feel stupid lol.

J-H-Freedman, 2024-09-07:
jumped the gun :(

```
a@pop-os:~$ sudo make install
[  9%] Building CXX object CMakeFiles/joycond.dir/src/main.cpp.o
[ 18%] Building CXX object CMakeFiles/joycond.dir/src/phys_ctlr.cpp.o
[ 27%] Building CXX object CMakeFiles/joycond.dir/src/virt_ctlr.cpp.o
[ 36%] Building CXX object CMakeFiles/joycond.dir/src/virt_ctlr_passthrough.cpp.o
[ 45%] Building CXX object CMakeFiles/joycond.dir/src/virt_ctlr_combined.cpp.o
[ 54%] Building CXX object CMakeFiles/joycond.dir/src/virt_ctlr_pro.cpp.o
[ 63%] Building CXX object CMakeFiles/joycond.dir/src/epoll_mgr.cpp.o
[ 72%] Building CXX object CMakeFiles/joycond.dir/src/epoll_subscriber.cpp.o
[ 81%] Building CXX object CMakeFiles/joycond.dir/src/ctlr_detector_udev.cpp.o
[ 90%] Building CXX object CMakeFiles/joycond.dir/src/ctlr_mgr.cpp.o
[100%] Linking CXX executable joycond
/usr/bin/ld: cannot open output file joycond: Is a directory
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/joycond.dir/build.make:241: joycond] Error 1
make[1]: *** [CMakeFiles/Makefile2:100: CMakeFiles/joycond.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
a@pop-os:~$ sudo systemctl enable --now joycond
Failed to enable unit: Unit file joycond.service does not exist.
a@pop-os:~$ sudo make install joycond
```

Joltemon, 2024-11-01:
> jumped the gun :(
>
> ```
> a@pop-os:~$ sudo make install
> [  9%] Building CXX object CMakeFiles/joycond.dir/src/main.cpp.o
> [ 18%] Building CXX object CMakeFiles/joycond.dir/src/phys_ctlr.cpp.o
> [ 27%] [...]

I'm getting this issue too
Did you manage to figure out howto fix it?

J-H-Freedman, 2024-11-19:
> I'm getting this issue too Did you manage to figure out howto fix it?

I did not :'(

colibris79, 2025-04-23:
Same problem here with Mint 21.3. Guess no solution? is this a too old kernel issue?