DanielOgorchock/joycond #150: Creating Combined Joy-Cons is Inconsistent and Takes a Long Time

Jazztache, 2024-10-04:
Hello!

I have this problem with `joycond` where even after spam-pressing ZL and ZR repeatedly, the controllers will not pair. Sometimes, the 'Combined Joy-Cons' controller will appear, but be unresponsive.

I have to wait for both joy-cons to have the P1 dot on, which confirms they are a combined controller. Problem is, getting to this point is very difficult, because even after pressed ZL and ZR, or L and R at the same time, holding both buttons, trying to hold them whilst pairing, it still takes quite a while before I can use them as a pair.

This happens on both Official Joy-Cons and a pair of Doyoky Epoch controllers. They both take no time at all to be recognised as individual controllers, but ages to finally pair them together for a full set of controls.

If you need any logs or system information, I am ready and willing to provide these if needed. Thankyou!

Jazztache, 2024-10-04:
To elaborate, the controllers consistently connects as Player 1 and Player 2, and for some reason I cannot seem to consistently get it to be combined joy-cons. I usually have to wait for a while.

Jazztache, 2024-10-07:
Update:
After stopping and starting the `joycond` daemon, and following the output with `journalctl --follow --lines 100 --unit joycond`, I have this:
```
Oct 07 21:55:54 arch-lenovo systemd[1]: Stopping joycond...
Oct 07 21:55:54 arch-lenovo systemd[1]: joycond.service: Deactivated successfully.
Oct 07 21:55:54 arch-lenovo systemd[1]: Stopped joycond.
Oct 07 21:56:20 arch-lenovo systemd[1]: Started joycond.
Oct 07 21:56:20 arch-lenovo joycond[83953]: adding epoll_subscriber: fd=4
Oct 07 21:56:31 arch-lenovo joycond[83953]: DEVNAME=event23 ACTION=add DEVPATH=/devices/virtual/misc/uhid/0005:057E:2006.0017/input/input75/event23
Oct 07 21:56:31 arch-lenovo joycond[83953]: Creating new phys_ctlr for /dev/input/event23
Oct 07 21:56:31 arch-lenovo joycond[83953]: Found Left Joy-Con
Oct 07 21:56:31 arch-lenovo joycond[83953]: driver_name: Joy-Con (L)
Oct 07 21:56:31 arch-lenovo joycond[83953]: MAC: A0:5A:5D:E3:DE:78
Oct 07 21:56:54 arch-lenovo joycond[83953]: adding epoll_subscriber: fd=5
Oct 07 21:56:54 arch-lenovo joycond[83953]: handle sync
Oct 07 21:56:54 arch-lenovo joycond[83953]: DEVNAME=event25 ACTION=add DEVPATH=/devices/virtual/misc/uhid/0005:057E:2007.0018/input/input77/event25
Oct 07 21:56:54 arch-lenovo joycond[83953]: Creating new phys_ctlr for /dev/input/event25
Oct 07 21:56:54 arch-lenovo joycond[83953]: Found Right Joy-Con
Oct 07 21:56:54 arch-lenovo joycond[83953]: driver_name: Joy-Con (R)
Oct 07 21:56:54 arch-lenovo joycond[83953]: MAC: A0:5A:5D:E3:E1:D3
Oct 07 21:57:08 arch-lenovo joycond[83953]: adding epoll_subscriber: fd=14
Oct 07 21:57:09 arch-lenovo joycond[83953]: handle sync
Oct 07 21:57:09 arch-lenovo joycond[83953]: handle sync
Oct 07 21:57:09 arch-lenovo joycond[83953]: Waiting controller needs partner
Oct 07 21:57:09 arch-lenovo joycond[83953]: Found left
Oct 07 21:57:09 arch-lenovo joycond[83953]: Waiting controller needs partner
Oct 07 21:57:09 arch-lenovo joycond[83953]: Found right
Oct 07 21:57:09 arch-lenovo joycond[83953]: adding epoll_subscriber: fd=25
```

Even after this, when it says it has found the controller, it doesn't work.
It has the `Nintendo Switch Combined Joy-Cons` in the KDE Game Controller Settings menu but it still doesn't take any input.

The LEDs often show different player numbers, or one rotates the LEDs whilst the other stays static.
This happens for a while before they both stop on the P1 LED after a decent bit of waiting, after which the controller finally works.

I even went back and checked and there was no new joycond output in `journalctl`.

I tried with both official Joy-Cons and the Binbok/Doyoky Epoch Joy-Cons. The Doyokys are noticably worse, so merging https://github.com/DanielOgorchock/joycond/pull/116 might alleiviate this issue.

Apologies for not providing enough information to start with, as I was not aware how to debug and troubleshoot this issue up until now.

KonstantinDjairo, 2025-01-22:
use this fork that has auto-pair

https://github.com/SeongGino/joycond

hanhha, 2025-04-15:
I got same issue. Looked like the led control took long time to complete. I commented out all code that control the led and it worked immediately.

BeeFox-sys, 2025-05-14:
Upon this, i had a look at the code and i think the actual issue is that the pairing lights pattern continues after the controllers are combined, my solution was just to comment out all the actionable code that enabled these pairing lights, lines [366](https://github.com/DanielOgorchock/joycond/blob/39d5728d41b70840342ddc116a59125b337fbde2/src/phys_ctlr.cpp#L366) to [375](https://github.com/DanielOgorchock/joycond/blob/39d5728d41b70840342ddc116a59125b337fbde2/src/phys_ctlr.cpp#L375) of phys_ctrl.cpp. my suspect is that joycond is failing to tell the ledtrig-timer module to stop blinking the lights or something like that, but i don't think i know the code base well enough to be certain.