DanielOgorchock/joycond #154: linux mint doesn't even start pairing

electikos, 2024-10-28:
like in the title the LED of my joycon don't blink like said in the tuto but continue to do pinpong.
does someone know how to pair the joycon in this case?

ParkerHGordon, 2025-02-25:
Looking to try and connect my Switch Pro controller on Linux Mint as well. Supposedly hid-nintendo is included in the latest versions of the linux kernel. I'm a linux noob, so maybe I'm misunderstanding what lsmod is supposed to do, but I don't see hid-nintendo included in the output. Did linux mint's version of the kernel remove hid-nintendo for some reason? Because of that's the case, that's probably why we can't connect.

ParkerHGordon, 2025-02-26:
Update: I was previously on linux kernel 6.8.0-54, so I tried updating to linux kernel 6.11.0-17 which includes some changes to both bluetooth and hid-nintendo. This actually fixed it, kinda! I can successfully pair my switch pro controller with my computer now by pressing and holding the pair button on the controller, connecting on the computer, then pressing L+R on my controller. Works with and without joycond running. 

The problem I still have is that whenever I disconnect the controller, I have to repair from scratch to get it to work. Still looking for a fix on that, but I wanted to include my progress in case it helps anyone!