DanielOgorchock/joycond #161: Does not enter pairing mode using third party pro controller

LittlePlanlessFool, 2025-03-11:
I am using a PowerA Fusion Pro Controller.
Connecting over Bluetooth, the player LEDs pingpong from left to right and back. Over USB, no lights at all. Pressing L+R, ZL+ZR and Start+Select does nothing either way.
output of `systemctl status joycond`:
```joycond.service - joycond
Loaded: loaded (/etc/systemd/system/joycond.service; enabled; preset: disabled)
Drop-In: /usr/lib/systemd/system/service.d
         10-timeout-abort.conf, 50-keep-warm.conf
Active: active (running) since Tue 2025-03-11 11:48:19 CET; 5min ago
Invocation: e6ab25b1c81f4b8f9746ed6213707a85
Main PID: 15680 (joycond)
Tasks: 1 (limit: 75855)
Memory: 380K (peak 1.3M)
CPU: 9ms
CGroup: /system.slice/joycond.service
         15680 /usr/bin/joycond

Mar 11 11:48:19 hostname systemd[1]: Started joycond.service - joycond.
Mar 11 11:48:19 hostname joycond[15680]: adding epoll_subscriber: fd=4```

ConfusedKeys, 2025-03-12:
I also have the PowerA Fusion Pro Controller. I have the same issue and same output from `systemctl status joycond` . It would be absolutely fantastic if this controller could work with joycond, but I can see that supporting third party controllers is not the main goal of this project.

LittlePlanlessFool, 2025-05-09:
Sadly, it did not.
However, installing [hid-nintendolic](https://gitlab.com/cipitaua/dkms-hid-nintendolic) did make steam register full deflection on both sticks when connected via usb, but bluetooth and the gyro still remain nonfunctional.