DanielOgorchock/joycond #162: Controllers are connected to bluetooth but pressing the triggers does nothing

zaxinothdigital, 2025-04-08:
I have checked evtest, and the devices are not listed. Nothing with the word "joycon" or anything. I am on Linux Mint 22 Cinnamon and my kernel is 6.8.0-51-generic. The service is started already.

askiiart, 2025-04-11:
Seconding this; got a 1st party pro controller working over USB, but same controller (or official joycons, for that matter) over bluetooth does not work. It just shows the blinky lights as if it's still in pairing mode. Nothing in dmesg or systemctl.

zaxinothdigital, 2025-04-11:
Update: I got the driver to work but now controllers are randomly disconnecting almost as soon as I connect the other one

askiiart, 2025-04-12:
I got it to work by first pairing the controller, then connecting it, then trusting it - better instructions [here](https://github.com/bluez/bluez/issues/673#issuecomment-1849132576).

That consistently worked for me, though sometimes after disconnecting it, I'd have to remove the controller and go through the process again rather than just reconnecting to it.

alexemanuel, 2025-05-24:
Thanks, @askiiart. That really worked to me